RobotLocomotion/drake #5718: Ensure that MathematicalProgram unittest problems inherit from test::OptimizationProblem when able

EricCousineau-TRI, 2017-04-04:
Presently, there are some `MathematicalProgram` unittest problems which do not leverage `drake::solvers::test::OptimizationProblem`.
One example is the `LinearSystemExample1` problem: [source](https://github.com/RobotLocomotion/drake/blob/4c8f868/drake/solvers/test/optimization_examples.h#L53).

It would be useful to consolidate these test problems to leverage the existing `OptimizationProblem` interface. We can consider then having `OptimizationProblem` inherit from `MathematicalProgram`.

This stems from the comments in #5685.

jwnimmer-tri, 2017-05-05:
All issues must have owners; feel free to reassign elsewhere as needed (or close), etc.

hongkai-dai, 2019-03-25:
@EricCousineau-TRI , is this issue still active? Feel free to close or reassign it.

EricCousineau-TRI, 2019-06-20:
Ehhh, I'm gonna close this one. Doesn't seem useful to keep open.

closed 2019-06-20 by EricCousineau-TRI