RobotLocomotion/drake #5725: "now" event systems::DiscreteEvent::ActionType

siyuanfeng-tri, 2017-04-05:
There are a number of instances in the code base (https://github.com/RobotLocomotion/drake/blob/d0a195880b7933d2ef9743279a788bcb25144fe3/drake/automotive/maliput_railcar.h#L70-L83, https://github.com/RobotLocomotion/drake/pull/5712, https://github.com/RobotLocomotion/drake/pull/5724) that want to schedule an event to be handled now now. However returning time = 0 in CalcNextUpdateTime causes some unhappiness for the integrator. The current workaround is to ask for an event scheduled at context.time() + epsilon. 
A better work around is to insert a bit of special logic to catch such events before the integrator around https://github.com/RobotLocomotion/drake/blob/master/drake/systems/analysis/simulator.h#L457

cc @jwnimmer-tri @sherm1 @edrumwri @liangfok

liangfok, 2017-04-05:
Here is the exception that is thrown when the next update time equals the current time:

https://github.com/RobotLocomotion/drake/blob/1c4dfede2abfc82611fdd04bf991cc29cc1659c3/drake/systems/analysis/integrator_base.h#L1284

jwnimmer-tri, 2017-04-05:
Here was one rough sketch of a solution from a couple days ago: https://github.com/jwnimmer-tri/drake/tree/simulator-zerotime-update

sherm1, 2017-04-05:
@edrumwri and I are considering the implications of allowing a zero-length continuous step (which is what happens with a "now" update time, because CalcNextUpdateTime() is necessarily called after discrete variables have been updated at step start). My past experience with these has been unpleasant because they easily lead to simulations that hang at seemingly-arbitrary times. 

An alternative would be a "discrete event" trigger that the Simulator could invoke at step start so that the handler could be invoked along with the other discrete changes that occur at the start of every step. But ... I like the simplicity of just allowing the 0-length step if it can work properly.

sherm1, 2018-08-13:
@edrumwri I believe this issue was fixed in #8486 -- can we close it now?

closed 2018-08-13 by edrumwri