RobotLocomotion/drake #5730: Should event handling callbacks have a System argument?

sherm1, 2017-04-05:
`drake::systems::DiscreteEvent` typically invokes one of the generic handler APIs for Publish, DiscreteUpdate, or UnrestrictedUpdate. However it also allows any subsystem to provide custom callbacks. The generic methods resolve to virtual member functions like DoPublish() and consequently have access to the System as implicit first argument. The signatures of the custom ones don't include a System argument (which is always needed) so supplying them requires code like this: 
```c++
      // Use a custom update action.
      event.action = DiscreteEvent<double>::kDiscreteUpdateAction;
      event.do_calc_discrete_variable_update = std::bind(
                                  &TestSystem::DoCalcDiscreteUpdatesNumber,
                                  this, std::placeholders::_1 /* context */,
                                  std::placeholders::_2 /* discrete state */,
                                  kNumberToUpdate);
```

I'm not looking forward to explaining that to normal humans! This would be simpler if the signature required a `const System*` as its first argument. Should it?

jwnimmer-tri, 2017-04-05:
If we instead suggest and provide examples using lambdas for the callbacks, perhaps that is easier on all fronts?

sherm1, 2017-04-05:
A lambda would make a nicer example. But in the test case I grabbed the update function is a member of the TestSystem class, which expects its first argument to be a `this` pointer. If that is a common case (which I think is likely since this methods can be complicated) it might be more helpful to show the complicated `std::bind` stuff for copy/paste purposes since users are likely to struggle with that. It would be simpler if the first argument were a `const System*` since the context and state arguments wouldn't have to be relocated.

But ... as-is the bound `this` pointer already has the concrete System type while with my proposal it would have the base type and have to be downcast to be useful. So maybe it is better as it is -- I'll close this. Thanks for your indulgence!

closed 2017-04-05 by sherm1