RobotLocomotion/drake #5742: Something seems odd with LCM-based kuka simulations

naveenoid, 2017-04-06:
Onwards from #5712  (change the lcm subscriber system to properly obey context in it's DoCalcOutput), the dev/pick_and_place_state_machine (LCM based pick and place demo) works a *lot* slower. I have not investigated further but the commit clearly does not affect standalone type demos. 

Assigned to @siyuanfeng-tri but also tagging @sammy-tri

siyuanfeng-tri, 2017-04-06:
confirmed.
the arm definitely moves much slower, but the lcm publish rate from the simulator seems to be normal. suggesting the simulation is running at normal speed. 
weird though. i will look into it tomororw.

siyuanfeng-tri, 2017-04-06:
naveen, if you have some spare cpu time, can you figure out where it broke exactly between my merge and sammys? thanks

naveenoid, 2017-04-06:
Actually Im diggin in now and I am not sure if its your PR or sammys..Either case the puzzling thing is that only the IIWA is significantly slower.

naveenoid, 2017-04-06:
I will investigate further and report by the time you are up again ;)

siyuanfeng-tri, 2017-04-06:
ok. i understand what's wrong now. it's a combination of the following artifacts:
1. kuka_plan_runner is sending command messages over lcm as soon as it receives a new status message from the simulation. 
2. simulation is configured to publish on every step, which sends a status message.
3. with the new lcm subscriber (https://github.com/RobotLocomotion/drake/pull/5712#event-1031007910), every time a new command message arrives, it takes a tiny little step so that the message can be processed and put in the context properly. however this tiny little step triggers a simulation publish, which triggers another command publish later, which triggers another tiny little step to process the message, and it goes on. 

i can fix it like this (pr in progress): 
1. rate limit kuka_plan_runner to a reasonable thing, like 200hz.
2. don't publish on every tick in simulation, rather at a fixed rate. this is not exactly a fix, but should be done anyway. 

however, this is a good lesson, it showed us that:
0. i forgot to test this case manually before i merged the pr...
1. we don't have test coverage for this complicate lcm driven sim + command source. and i am not sure how to test this exactly. suggestions are welcomed. 
2. potentially, if you are publishing lcm message "faster than you really needs", it is possible to cause the simulation to slow down tremendously. this was not possible before. 
3. maybe for all such applications, the design guide line should be always publish at a fixed rate? that match hardware anyway.

david-german-tri, 2017-04-06:
> we don't have test coverage for this complicate lcm driven sim + command source. and i am not sure how to test this exactly. suggestions are welcomed.

Is [DrakeMockLcm](https://github.com/RobotLocomotion/drake/blob/master/drake/lcm/drake_mock_lcm.h) helpful?  (example usage)[https://github.com/RobotLocomotion/drake/blob/4532d9460627535e5feb7d0f19ce951b35b19f95/drake/multibody/rigid_body_plant/test/rigid_body_plant_that_publishes_xdot_test.cc#L40]

siyuanfeng-tri, 2017-04-06:
@david-german-tri i don't think so, the premise here is things are in different processes., last time i looked at the mock lcm thing, it's in memory loopback right?

david-german-tri, 2017-04-06:
> last time i looked at the mock lcm thing, it's in memory loopback right?

Yes, so it might be useful for unit tests of the code that publishes messages, for instance to lock in that you are publishing at a fixed rate.

siyuanfeng-tri, 2017-04-06:
ah, i see. makes sense.

closed 2017-04-11 by siyuanfeng-tri