RobotLocomotion/drake #5744: can't build master on mac (w/ bazel)

RussTedrake, 2017-04-06:
just tried verifying something quickly on mac, but it seems like the snopt bindings have broken?
```
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: @snopt//csrc:snopt_c: no such attribute 'strip_include_prefix' in 'cc_library' rule.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:np02lib.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:npopt.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn02lib.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn03prnt.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn05wrpa.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn05wrpb.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn05wrpc.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn05wrpn.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn10mach.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn12ampl.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn15blas.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn17util.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn20amat.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn25bfac.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn27lu.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn30spec.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn35mps.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn37wrap.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn40bfil.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn50lp.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn55qp.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn56qncg.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn57qopt.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn60srch.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn65rmod.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn70nobj.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn80ncon.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn85hess.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn87sopt.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn90lmqn.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sn95fmqn.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:snfilewrapper.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:snopta.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:snoptb.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:snoptc.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:snoptq.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:sq02lib.c' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
[... cut ...]
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/csrc/BUILD:3:1: Target '@snopt//csrc:snopt.hh' contains an error and its package is in error and referenced by '@snopt//csrc:snopt_c'.
ERROR: /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/snopt/BUILD:5:1: Target '@snopt//csrc:snopt_c' contains an error and its package is in error and referenced by '@snopt//:snopt_c'.
ERROR: Analysis of target '//drake/solvers:mathematical_program' failed; build aborted.
```

jwnimmer-tri, 2017-04-06:
> external/snopt/csrc/BUILD:3:1: @snopt//csrc:snopt_c:
> no such attribute 'strip_include_prefix' in 'cc_library' rule.

The `strip_include_prefix` is a feature added to Bazel as of 0.4.3 (which is our required version).  Could you check what version of Bazel you are using?

jamiesnape, 2017-04-06:
Is there a way to do a version check at the start of the Bazel build?

soonho-tri, 2017-04-06:
This commit is related: https://github.com/bazelbuild/bazel/commit/6c2aaacb5b9b01fa5614b6a2b147859fc70c9125

jwnimmer-tri, 2017-04-06:
Sounds like a great idea.

david-german-tri, 2017-04-06:
Nice - another reason to bump to 0.4.5.

RussTedrake, 2017-04-06:
doh.  was on 0.4.2 .  too many machines.  thanks all!

now I get:
```
ERROR: /Users/russt/drake-distro/tools/BUILD:38:1: no such package '@com_github_bazelbuild_buildifier//buildifier': Error downloading [https://github.com/bazelbuild/buildifier/archive/93929369232fcda305607a2e0aa7b3cd9cf8912d.tar.gz] to /private/var/tmp/_bazel_russt/3e477aa1673e9c80e91d410bb65d1970/external/com_github_bazelbuild_buildifier/93929369232fcda305607a2e0aa7b3cd9cf8912d.tar.gz: Checksum was 2ffb39756767165133f9861d8bf52c76d5474bb462583edfe47d2ea0a759c62b but wanted 51a84592ced38213592a2d177cf551d3be9179092d3921a38154a2662610df96 and referenced by '//tools:buildifier'.
ERROR: Analysis of target '//tools:buildifier' failed; build aborted.
```

jwnimmer-tri, 2017-04-06:
I don't have the issue number atop my head, but sadly Bazel caches (or at least used to cache) download failure messages.  So if you fail once, it will remember the failure and not retry (unless you change something about the URL or etc).  The work-around is to change the URL and retry (and then put it back), or else `bazel clean`.

soonho-tri, 2017-04-06:
@jwnimmer-tri , I think we do have a wrong `sha256` at https://github.com/RobotLocomotion/drake/blob/master/WORKSPACE#L209.

soonho-tri, 2017-04-06:
I can reproduce the problem by running:
```bash
bazel clean
./tools/buildifier.sh
```

jwnimmer-tri, 2017-04-06:
OH!  They renamed their repository an hour ago.  Ugh!

jwnimmer-tri, 2017-04-06:
That's a more satisfying answer (because I thought download-fail caching was already fixed), but annoying that bisect is broken forever in the past now.

Yet another reason to get Drake-specific mirroring of tarballs up into our CI.

soonho-tri, 2017-04-06:
lol, that solves the mystery. I was thinking that we're under a man-in-the-middle attack..

PR #5757 (2017-04-06): Update buildifier repo name and sha256
Close #5744

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5757)
closed 2017-04-06 by soonho-tri