RobotLocomotion/drake #5745: Request to remove lane length restriction in DrawLaneArrow

basicNew, 2017-04-06:
There is a restriction in [DrawLaneArrow](https://github.com/RobotLocomotion/drake/blob/master/drake/automotive/maliput/utility/generate_obj.cc#L413) on generate_obj.cc, which requires the lane to have a minimum length (from what I've seen this is to make sure you can fit the two triangles used to depict the lane direction). This restriction becomes quite difficult to meet when dealing with arbitrary road network geometries, so I was wondering if it could be removed. Below there are two examples that I bumped into and I think should be valid, but fail to pass that assertion (note: I generated the obj files for these by commenting out that `DRAKE_DEMAND` line):

[image]

[image]

liangfok, 2017-04-06:
I agree also because the arrows might confuse the viewer into thinking they imply a direction of vehicle travel. Perhaps they can be a feature that is by default disabled but optionally enabled.

basicNew, 2017-04-06:
Glad that you mentioned it @liangfok, I actually though that the arrows depict the lane direction (in the sense that cars should traverse them in that way).

liangfok, 2017-04-06:
Another idea is to place a single arrow in the middle of the segment with no restriction on its minimum length (meaning it could appear like a line when the segment is very short), but include some other markers to denote segment boundaries.

maddog-tri, 2017-04-07:
@basicNew, could you please attach a test case that exercises this failure?  (E.g., a small monolane YAML file that triggers it.)

basicNew, 2017-04-07:
Sure. This monolane:

```
# -*- yaml -*-
---
# distances are meters; angles are degrees.
maliput_monolane_builder:
  id: city_1
  lane_bounds: [-2, 2]
  driveable_bounds: [-4, 4]
  position_precision: 0.01
  orientation_precision: 0.5
  points:
    street_9_1_3:
      xypoint: [92.92893218813452, 307.0710678118655, -45.0]
      zpoint: [0.0, 0, 0, 0]
    street_9_1_4:
      xypoint: [95.05025253169417, 304.9497474683058, -45.0]
      zpoint: [0.0, 0, 0, 0]
  connections:
    street_9_1_3-street_9_1_4: {start: points.street_9_1_3, length: 3.000000000000014,
      explicit_end: points.street_9_1_4}
  groups: {}
```
gives me this error:

```
$ yaml_to_obj -yaml_file="generated_worlds/city_monolane.yaml" -obj_dir="generated_worlds/" -obj_file="city_monolane"
[2017-04-07 09:23:04.536] [console] [info] Loading road geometry.
[2017-04-07 09:23:04.538] [console] [info] Generating OBJ.
abort: failure at drake/automotive/maliput/utility/generate_obj.cc:413 in DrawLaneArrow(): assertion '(s_offset + s_size) <= lane->length()' failed.
```

If I comment out that line, I get this segment generated:

[image]

maddog-tri, 2017-04-07:
Thanks for the test case!  It has been incorporated into a unit-test in the pending PR.

FYI, the arrows can already be disabled --- see `maliput::utility::ObjFeatures.draw_arrows`.  (The default is `true`.  I have found them really handy for debugging/understanding the road geometry.)

closed 2017-04-07 by maddog-tri

basicNew, 2017-04-08:
@maddog-tri thanks for the quick fix and hint on disabling arrows!

PR #5768 (2017-04-07): Fixes #5745 ("Request to remove lane length restriction in DrawLaneArrow").
Fixes #5745 ("Request to remove lane length restriction in DrawLaneArrow").

The "lane length restriction" is actually an assertion ensuring that
`DrawLaneArrow()` is not being asked to draw an arrow which pokes out beyond
the end of the lane.   The caller (`MarkLaneEnds()`) does in fact try to
calculate parameters which avoid that --- but apparently numeric precision
issues in `double` arithmetic can cause the assertion to be violated.

Solution:  nudge arrows inwards by `linear_tolerance()` of the `RoadGeometry`,
whi [... cut]
commit 81d224a9 (2017-04-07): Fixes #5745 ("Request to remove lane length restriction in DrawLaneArrow"). (#5768)

The "lane length restriction" is actually an assertion ensuring that
`DrawLaneArrow()` is not being asked to draw an arrow which pokes out beyond
the end of the lane.   The caller (`MarkLaneEnds()`) does in fact try to
calculate parameters which avoid that --- but apparently numeric precision
issues in double arithmetic can cause the assertion to be violated.

Solution:  nudge arrows inwards by `linear_tolerance()` of the RoadGeometry,
which provides plenty of buffer to avoid precision issues.

Also:  create a [... cut]