RobotLocomotion/drake #5746: show time in drake_visualizer?

siyuanfeng-tri, 2017-04-06:
@patmarion how hard is to render sim time in the visualizer? would be super useful.

patmarion, 2017-04-06:
There are several examples of how to do this, would be happy to show you.  If you work on this, it would be great if you could do a complete implementation (show sim time and real time factor, possibly other sim stats), but you can also make a quick n dirty implementation just by pasting a few lines of Python.

siyuanfeng-tri, 2017-04-06:
sure i can do it. maybe next week when i am back to cam?

naveenoid, 2017-04-06:
Oh sooooo want this!

patmarion, 2017-04-06:
This is the minimal example that you can just copy and paste this into the `drake-visualizer` Python console:

```
from director import lcmUtils
import bot_core as lcmbotcore

def onViewerDraw(msg):
    t = msg.timestamp*1e-3
    vis.updateText('sim time: %.3f' % t, 'sim time')

lcmUtils.addSubscriber('DRAKE_VIEWER_DRAW', lcmbotcore.viewer_draw_t, onViewerDraw)
```

A more complete implementation might define a class that manages the subscriber and provides an API to query for information like current sim time, sim real time factor, etc, and a separate UI class that displays the information in a panel or in the status bar with a menu option to toggle the feature on/off.

siyuanfeng-tri, 2017-04-06:
this is all i want, is there a way to avoid typing this every time in the viz?

siyuanfeng-tri, 2017-04-06:
that probably requires compiling some c code on your end?

patmarion, 2017-04-06:
`drake-visualizer --script myscript.py`

But please consider investing the time to contribute this feature to `drake-visualizer` in a way that everyone can benefit from it.  My previous comment described what I would consider to be a welcome contribution.

siyuanfeng-tri, 2017-04-06:
would be next week then.

patmarion, 2017-04-06:
While we're on the topic of sim time, I use this script to replay sim lcm logs at a realtime rate:

https://github.com/RobotLocomotion/spartan/blob/master/scripts/sim_playback.py

`drake-visualizer --script sim_playback.py simlog.lcmlog`

closed 2017-09-18 by siyuanfeng-tri