RobotLocomotion/drake #5750: AutomotiveSimulator Should Delete Temporary Modeling Files After They Are No Longer Needed

liangfok, 2017-04-06:
See: https://reviewable.io/reviews/RobotLocomotion/drake/5741#-Kh3DAHYsSPCGMDxzaMg

liangfok, 2017-04-06:
Since the modeling files were created using `maliput::utility::GenerateUrdfFile()`, I think it may be best to provide a sister `maliput::utility::RemoveGeneratedUrdfFile()` that undoes the filesystem artifacts.

jwnimmer-tri, 2017-04-06:
A better answer is to create them somewhere that is automatically cleaned up, such as the Bazel sandbox, a system location that has a janitor daemon, and/or a transient filesystem.

Having programmers match create / destroy function calls manually will never work.

liangfok, 2017-04-06:
Thanks. I've updated the title accordingly.

liangfok, 2017-04-12:
I'm going to close this issue because the files are currently created in `/tmp`, which is automatically cleared on our target platforms.

I briefly looked into using Bazel's `genrule()` to generate the road network files but figured I can't without explicitly enumerating and thus limiting the types of `maliput::api::RoadGeometry` objects supported. Currently, _any_ `RoadGeometry` can be passed to `AutomotiveSimulator::SetRoadGeometry()`, which then generates the URDF, OBJ, and MTL files in `/tmp`.

closed 2017-04-12 by liangfok