RobotLocomotion/drake #5777: How to detect contact force?

GuiyangXin, 2017-04-09:
Hello guys,
I want to detect the contact forces on feet for LittleDog. I found a example named as contactSensorTest.m. The method is to use "ContactForceTorqueSensor" class and the "addSensor". My question is the same as in this Issue [https://github.com/RobotLocomotion/drake/issues/102](url). I don't know how to understand the result because I don't know the relationship between the sensor's frame and the world frame. I want to get the result for contact force in world frame. The code to add sensor is as follows,
`r = addSensor(r,FullStateFeedbackSensor());
body = findLinkId(r,'front_left_lower_leg');
frame = RigidBodyFrame(body,zeros(3,1),zeros(3,1),'FT_frame');
r = addFrame(r,frame);
r = addSensor(r,ContactForceTorqueSensor(r,frame));
r = compile(r);`

hongkai-dai, 2017-04-11:
The line
```MATLAB
frame = RigidBodyFrame(body,zeros(3,1),zeros(3,1),'FT_frame');
```
says how the frame is attached  to the body `front_left_lower_leg`. The first `zeros(3,1)` is the position of the `FT_frame` in the body frame, and the second `zeros(3,1)` is the orientation of the `FT_frame` in the body frame (using Euler angles in space-fixed x-y-z order). In this case, the force torque sensor frame `FT_frame` aligns with the body frame `front_left_lower_leg`, since both the position and orientation are zero. 

In order to get the orientation of the `FT_frame` in the world frame, for a given robot posture `q`, you need to do
```MATLAB
kinsol = r.doKinematics(q);
pose = r.forwardKin(kinsol, frame, zeros(3,1), struct('rotation_type', 2));
```
Now `pose(1:3)` is the position of the force torque sensor in the world frame. and `pose(4:7)` is the quaternion of the force torque sensor in the world frame.

GuiyangXin, 2017-04-15:
Thank you Hongkai for your reply. However, I encountered other problems. First,  according to my code, particularly `frame = RigidBodyFrame(body,zeros(3,1),zeros(3,1),'FT_frame');`, I have put the force sensor on the origin of the body frame. Should I put it on the surface of foot in order to detect contact force? I don't know the position of the end point of `front_left_lower_leg`. But I know the foot center according to the definition of `front_left_foot_center` frame in LittleDog's urdf file. Therefore, I had a try to put the force sensor on the center of foot using `frame = RigidBodyFrame(body,[-0.0265 0 -0.0985]',zeros(3,1),'FT_frame');`. What's interesting is that the outputs of force-torque sensors in these two cases are identical. I think the forces should be identical, but moments will change depending on different position of force-torque sensor. So it is a little bit weird. 
Another question is why I cannot use Matlab function `eul2rotm` to convert Euler angles to rotation matrix. I found there is a function named as `quat2rotmat`  to convert quaternion to rotation matrix whereas the same function in Matlab is `quat2rotm`. Does it mean I cannot use some functions of Matlab when Drake is running? Thanks in advance.

hongkai-dai, 2017-04-15:
> Should I put it on the surface of foot in order to detect contact force?

No you do not have to. As long as the force torque sensor is rigidly attached to the foot body frame, then you should be able to transform the force/torque reading in the sensor frame, to the foot frame.

>  What's interesting is that the outputs of force-torque sensors in these two cases are identical. I think the forces should be identical, but moments will change depending on different position of force-torque sensor. So it is a little bit weird.

That is weired to me as well. How do you get the force torque sensor reading in Drake? Could you paste your code here?

> Does it mean I cannot use some functions of Matlab when Drake is running?

Yes you can use Matlab functions such as `eul2rotm` and `quat2rotm`. The reason why we use `Euler2rotmat` and `quat2rotmat` is to also return the gradient and hessian of the function. The gradient and Hessian are useful in gradient-based nonlinear optimization.

GuiyangXin, 2017-04-15:
Thank you very much. My code is as follows. You need to run this program twice. First time please use `frame = RigidBodyFrame(body,zeros(3,1),zeros(3,1),'FT_frame');`. Second time please use `frame = RigidBodyFrame(body,[-0.0265 0 -0.0985]',zeros(3,1),'FT_frame');`. Please check it.
`r = LittleDog;
sys = TimeSteppingRigidBodyManipulator(r,0.001);
sys = addSensor(sys,FullStateFeedbackSensor());
body = findLinkId(sys,'front_right_lower_leg');

% Put force sensor on different position of the front_right_lower_leg
% frame = RigidBodyFrame(body,zeros(3,1),zeros(3,1),'FT_frame');
frame = RigidBodyFrame(body,[-0.0265 0 -0.0985]',zeros(3,1),'FT_frame');

sys = addFrame(sys,frame);
sys = addSensor(sys,ContactForceTorqueSensor(sys,frame));
sys = compile(sys);
v = constructVisualizer(sys);
sys1=sys;

foot = struct('id',[],'in_stance',[]);
foot(1).id = sys.findFrameId('front_left_foot_center');
foot(2).id = sys.findFrameId('front_right_foot_center');
foot(3).id = sys.findFrameId('back_left_foot_center');
foot(4).id = sys.findFrameId('back_right_foot_center');
body_id=findLinkId(sys,'body');
FT_frame_id=sys.findFrameId('FT_frame');
    
x0 = home(r);
qa0 = x0(getActuatedJoints(r));
nq = r.getNumPositions();
qstar = x0(1:nq);
v.draw(0,qstar);

hip_roll = 0.1;
hip_pitch = 1;
knee = 1.55;
xstart = Point(getStateFrame(sys));
xstart.front_right_hip_roll = -hip_roll;
xstart.front_right_hip_pitch = hip_pitch;
xstart.front_right_knee = -knee;
xstart.front_left_hip_roll = hip_roll;
xstart.front_left_hip_pitch = hip_pitch;
xstart.front_left_knee = -knee;
xstart.back_right_hip_roll = -hip_roll;
xstart.back_right_hip_pitch = -hip_pitch;
xstart.back_right_knee = knee;
xstart.back_left_hip_roll = hip_roll;
xstart.back_left_hip_pitch = -hip_pitch;
xstart.back_left_knee = knee;
xstart.base_z = 0.146;

T=2;
      
% construct PD control 
Kp = 5*eye(12);
Kd = diag([0.5; 0.6; 0.16; 0.5; 0.6; 0.16; 0.5; 0.5; 0.16; 0.5; 0.5; 0.16]);
sys = pdcontrol(sys,Kp,Kd);
      
% send position reference
sys = cascade(setOutputFrame(ConstOrPassthroughSystem(qa0),getInputFrame(sys)),sys);

if (0)
    sys = cascade(sys,v);
    simulate(sys,[0 2],double(xstart));
else
    [xtraj,ytraj] = simulate(sys,[0 2],double(xstart));
    playback(v,xtraj,struct('slider',true));
    figure
    plot(xtraj.xx(37,:))
    figure
    plot(xtraj.xx(38,:))
    figure
    plot(xtraj.xx(39,:))
    figure
    plot(xtraj.xx(40,:))
    figure
    plot(xtraj.xx(41,:))
    figure
    plot(xtraj.xx(42,:))
end

% Convert the result into world frame
options.rotation_type=2;
for i=2:T/0.001+1
    kinsol = doKinematics(sys1,xtraj.xx(1:18,i-1));
    pose=zeros(3,4);
    pose= forwardKin(sys1,kinsol,FT_frame_id,[0;0;0],options);
    quat = pose(4:7);
    rotm = quat2rotmat(quat);
    force(:,i-1)=rotm*xtraj.xx(37:39,i-1);
end
figure
plot(force(1,:))
figure
plot(force(2,:))
figure
plot(force(3,:))`

GuiyangXin, 2017-04-15:
> Yes you can use Matlab functions such as eul2rotm and quat2rotm. The reason why we use Euler2rotmat and quat2rotmat is to also return the gradient and hessian of the function. The gradient and Hessian are useful in gradient-based nonlinear optimization.

If I use `quat2rotm` in my program, it will be prompted an error `Undefined function 'quat2rotm' for input arguments of type 'double'.`

GuiyangXin, 2017-04-19:
Hello @hongkai-dai , I think the measurement result is right because 4 times the contact force element of z axis is equal to the gravity force of LittleDog.  But I have another question about the fiction coefficient between the feet and terrain. Can I set this coefficient? And how to do it? Thank you.

hongkai-dai, 2017-04-20:
Sorry I forgot to reply previously. For the friction coefficient. You can check contactConstraints.m, line 65. The friction coefficient is hard coded there.

GuiyangXin, 2017-04-20:
>  For the friction coefficient. You can check contactConstraints.m, line 65. The friction coefficient is hard coded there.

@hongkai-dai what I am trying to do is to push LIttleDog. As a result, when the terrain is very slippy, i.e. the friction coefficient is very little, the robot will slip on the ground rather than fall over. According to your comment, I have had a try to change the variable `mu = ones(nC,1);` in line 65 of contactConstraints.m as `mu = 0.1*ones(nC,1);.` But it doesn't work. When I define the class of LittleDog, I use Bullet. The code is as follows.
```
classdef LittleDog < TimeSteppingRigidBodyManipulator %& LeggedRobot
  
  methods
    function obj = LittleDog(options)
      options.floating = true;
      options.terrain = RigidBodyFlatTerrain();
      options.use_bullet=true;
      options.ignore_friction=false;
      options.ignore_self_collisions=false;
      options.dt=0.001;
      options.multiple_contacts=false;
      options.enable_fastqp=true;
      options.ignore_effort_limits = false;
      if ~isfield(options,'external_force')
        options.external_force = [];
      end
      w = warning('off','Drake:RigidBodyManipulator:UnsupportedContactPoints');
      obj = obj@TimeSteppingRigidBodyManipulator('LittleDog.urdf',options.dt,options);
      warning(w);
      % Add a force on a specified link if we want!
      if ~isempty(options.external_force)
        obj.external_force = findLinkId(obj,options.external_force);
        options_ef.weld_to_link = obj.external_force;
        obj = obj.addRobotFromURDF(fullfile(getDrakePath,'util','Copy_of_three_dof_force.urdf'), ...
          [0; 0; 0], [0; 0; 0], options_ef);
      end
    end
```
I suspect that the friction coefficient between feet and terrain depends on some parameters set up in the physical engine. But it is just a guess. Still need your help. Thank you very much.

hongkai-dai, 2017-04-20:
We do not use Bullet as a physics engine. Instead, we write the LCP solver to compute the state in the next control tick, as you can see in function solveLCP in TimeSteppingRigidBodyManipulator.m. Bullet is used for collision detection.

Could you add a break point in line 270 and line 420 of TimeSteppingRigidBodyManipulator, run the simulation, and let me know which breakpoint is hit? You can inspect the value of the friction coefficient `mu` then.

GuiyangXin, 2017-04-20:
I have a try. When running the simulation, it stopped at line 270 and didn't stop at line 420 of TimeSteppingRigidBodyManipulator. And I found that `mu` of the results `[phiC,normal,d,xA,xB,idxA,idxB,mu,n,D] = obj.manip.contactConstraints(kinsol, obj.multiple_contacts);` in this function `solveMexLCP(obj, t, x, u)` has changed if I change `mu = ones(nC,1)` in line 65 of contactConstraints.m.

hongkai-dai, 2017-04-20:
I am curious about your controller, you use a PD controller on the robot, with gains to be a diagonal matrix. But for a floating base robot as Little dog, its floating base is un-actuated, what's the meaning of using PD controller on the joints corresponding to the floating base? I think you are treating the robot as fully actuated, then I am not surprised that the little dog does not slide on the slippery surface, since the fully actuated robot does not need to rely on the friction force to stabilize itself.

GuiyangXin, 2017-04-21:
>  what's the meaning of using PD controller on the joints corresponding to the floating base?

The DOFs of LittleDog are composed of 6 floating base DOFs and 12 actuated joint DOFs. My PD controller is `Kp = 5*eye(12); Kd = diag([0.5; 0.6; 0.16; 0.5; 0.6; 0.16; 0.5; 0.5; 0.16; 0.5; 0.5; 0.16]); sys = pdcontrol(sys,Kp,Kd);`. It just controls the 12 joint DOFs except the 6 floating base DOFs.

hongkai-dai, 2017-04-24:
Have you tried to set the friction coefficient to 0 in contactConstraints, and then command the robot to move forward? Then I would expect the robot not to move, because there cannot be any horizontal force.

GuiyangXin, 2017-04-26:
Thank you Hongkai. I haven't let LittleDog walk or run. I am just working on the body motion control when the robot is standing on the ground using four legs. At this case, the simulation system doesn't show any slippy phenomenon even I set the friction coefficient to 0. I will try to make the robot walk and run next step. By the way, why is there a gap between LittleDog's feet and the ground viewed from the image in Director. For Atlas, the feet and the ground are combined without visible gap.

hongkai-dai, 2017-04-29:
That is probably a mismatch between the collision geometry and the meshes. You can try the following code
```MATLAB
r = RigidBodyManipulator('LittleDog.urdf', struct('floating','quat')); % Construct a robot whose floating base orientation is represented using quaternion.
v = r.constructVisualizer(); % Visualize the mesh geometry
v = r.constructVisualizer(struct('use_collision_geometry', true)); % Visualize collision geometry
```
I can see that the collision geometry has longer legs than the mesh geometry. There is probably some incorrect number in the URDF file. Actually when I searched for "collision" in the LittleDog.urdf, the documentation for the origin of the collision geometry says "it is approximate". I think these approximation is not quite right.

GuiyangXin, 2017-04-29:
Yes, I see the collision part contacts with terrain after adding your proposed code. So the simulation results is right even the robot looks like suspended in the air. All right, but I still need to figure out why my LittleDog cannot show any slipping feature when impacted by external force. Thanks again.

hongkai-dai, 2017-05-20:
Solved in #6123

closed 2017-05-20 by hongkai-dai