RobotLocomotion/drake #5781: Upgrade to Bazel 0.4.5

david-german-tri, 2017-04-10:
[Bazel 0.4.5](https://github.com/bazelbuild/bazel/releases/tag/0.4.5) provides some useful stuff, especially the `environ` parameter to `repository_rule`, which resolves [this TODO in gurobi.bzl](https://github.com/RobotLocomotion/drake/blob/master/tools/gurobi.bzl#L8).  

I believe #5762 fixed our last 0.4.5 backwards-compatibility issue - I've been beta-testing 0.4.5 on Linux for the past week with no problems, and @soonho-tri says it's working on Mac too.  So, let's upgrade the CI images, Xenial script, and documentation.

BetsyMcPhail, 2017-04-12:
CI images (Trusty, Xenial and OS X) have been updated. 

Documentation and Unprovisioned script updates coming shortly.

david-german-tri, 2017-04-12:
Fixed in #5808 - thanks!

closed 2017-04-12 by david-german-tri