RobotLocomotion/drake #5782: Consider setting `LCM_DEFAULT_URL` to a non-default URL for bazel tests (such as `memq://`)

EricCousineau-TRI, 2017-04-10:
When running unittests, `bazel` scrubs the environment variables, and therefore will use the default URL when executing `bazel test ...` ([example](https://github.com/EricCousineau-TRI/repro/blob/f78d33a/env_behavior/output.txt). This is generally not an issue, unless you are running `drake-visualizer` using the default `LCM_DEFAULT_URL`, where you will see the test artifacts appear.

It may be useful to use [`memq://`](https://lcm-proj.github.io/group__LcmC__lcm__t.html#gaf29963ef43edadf45296d5ad82c18d4b) if no IPC is required by the test.

david-german-tri, 2017-04-10:
Do you have a working prototype for this?  I haven't tried to coerce particular environment variables into `bazel test` before, but https://bazel.build/versions/master/docs/test-encyclopedia.html seems to suggest it isn't possible.

EricCousineau-TRI, 2017-04-10:
[`--action_env`](https://bazel.build/designs/2016/06/21/environment.html) seems like it can provide this, but it may only be updated upon a `bazel clean`, and does not seem to provide any warnings when it isn't provided by the current workspace:
https://github.com/EricCousineau-TRI/repro/blob/7a938c7/env_behavior/exec.sh

Here is the example output:
https://github.com/EricCousineau-TRI/repro/blob/7a938c7/env_behavior/output.txt#L4

jwnimmer-tri, 2017-04-10:
Most unit tests should be using mock LCM, not real LCM, anyway.  Maybe the (~3?) unit tests that need to send real LCM could call some programmatic API to change the URL, instead?

sammy-tri, 2017-04-10:
> Most unit tests should be using mock LCM, not real LCM, anyway. Maybe the (~3?) unit tests that need to send real LCM could call some programmatic API to change the URL, instead?

FWIW, there are some examples/ where someone may want to run it manually and watch the output.

jwnimmer-tri, 2017-04-10:
Sure, of course.  I thought those were out of scope with respect to the original posting.

EricCousineau-TRI, 2017-04-11:
@sammy-tri (Just wanted to echo our conversation form yesterday) For the unittests that you wish to run and inspect the output, you can use `bazel run` to directly execute the binary without an isolated environment; checked this by running `bazel run :main` on the above example.

@jwnimmer-tri That makes sense. Is there an easy way to figure out which tests are not using the mock LCM?

jwnimmer-tri, 2017-04-11:
@EricCousineau-TRI `drake/lcm/drake_mock_lcm.h` is the mock object; `drake/lcm/drake_lcm.h` is the real (transmitting / receiving) object.  You can find programs using one or the other using grep (and doing in the transitive-use reasoning yourself) or via `bazel query`.

jwnimmer-tri, 2017-05-05:
All issues must have owners; feel free to reassign elsewhere as needed (or close), etc.

EricCousineau-TRI, 2018-12-08:
Dusting this off, as it's been a small thing nagging at the back of my mind.

While this doesn't seem to be actively causing pain, we do still have instances where this happens, especially with places where we use `add_test_rule`.

For example, `//examples/manipulation_station:proof_of_life_test` still sends out LCM visualization messages, even when running under `bazel test`.

The manual fix is to add arguments to fix the environment variables; alternatively, we can use `--test_env` to inject an LCM url. We could possibly due this a little more easily, now that `install_prereqs` drops in a `.bazelrc` file.

EricCousineau-TRI, 2018-12-08:
I think it was finicky last time I tried when I tried it way back when, probably an invalid statement now.

I think `--action_env` would end up affecting how things are built (thus may require a rebuild), whereas `--test_env` might not? (Testing those theories now, will update this with the results.)

PR #10172 (2018-12-10): bazelrc: Prevent LCM messages from leaking out of tests
Resolves #5782 

This may be less than ideal, because `memq://` will prevent multiple LCM instances from talking to each other, thus someone will get mysterious failures when using `bazel test` vs. `bazel run`, so it'd be crappy to make this happen without some sort of announcement or whatever.

Alternatively, we could just try to obfuscate the `LCM_DEFAULT_URL` to something non-default, but that means we may still have tests collide or leak their information out.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion [... cut]
closed 2018-12-10 by EricCousineau-TRI