RobotLocomotion/drake #5785: Establish Eigen 3.3.3 as the One True Version

jwnimmer-tri, 2017-04-10:
Drake should switch to using Eigen 3.3.3 (the latest release) as its preferred version of Eigen, and document that changing it to be otherwise would be expensive -- because bleeding-edge Eigen becomes difficult to integrate into other projects that link in Drake.  We should no longer fork Eigen for ourselves, and bugs should be worked-around (and reported / fixed upstream) but not immediately bumped to the latest Eigen commit to receive them.

jamiesnape, 2017-04-11:
@m-chaturvedi has a pull request into Eigen that may be of interest before fixing a version: https://bitbucket.org/eigen/eigen/pull-requests/308/auto-diff-scalar-bug/diff

jwnimmer-tri, 2017-04-11:
I don't fully understand that issue yet maybe, but my initial reaction is that its the kind of thing that Drake should work around, and not need to bump the Eigen commit to fix.

jamiesnape, 2017-04-11:
It is bona fide memory issue, so I would disagree, but it is up to you.

jwnimmer-tri, 2017-04-11:
Drake could work around it by specializing `numeric_limits` for `AutoDiffScalar`, I think?

jamiesnape, 2017-04-11:
Yes, that works for me.

david-german-tri, 2017-04-11:
@stonier, this is relevant to our f2f conversation this morning.

jwnimmer-tri, 2017-04-13:
For more context on the rationale...

Eigen is a widely-used library in robotics and beyond (including for many internal projects within TRI).  Other projects will want to link Drake along with their choice of Eigen.  If Drake demands a bleeding-edge version of Eigen, then everyone else will have to eat that bleeding version also.  If Drake's minimum version of Eigen is more generous, then the other projects will have more flexibility in which Eigen to use.  And if Drake keeps bumping is version requirement over and over again, users will have to upgrade their tree to follow suit, which is especially annoying with other libraries might be integrated in binary form (e.g., PCL), or when newer Eigen imposes new requirements on users (see, e.g., the Eigen 3.3 upgrade notes, or its new `static_asserts` that defeat prior-working code).

Drake's use of Eigen in `MathematicalProgram` and `System<AutoDiffXd>` APIs as already fairly bleeding-edge, and we keep pushing up to newer commits and we want this or that little goodie.  My intent here is to stop the bleeding -- 3.3.3 is the newest we should use, and we find a way to make it work for whatever little sugar things we want.  Conceivably we could also _reduce_ the minimum required version to something earlier for even wider compatibility in the future, but for now at least I don't want to bump us into upgrades unless there is truly no other choice.

PR #5810 (2017-04-13): Switch Bazel to use Eigen 3.3.3
Closes #5785.

Note that this does _not_ switch the CMake version, because hopefully the CMake submodule goes away soon, and I don't see any information about how to update our `eigen-mirror`.  If someone else wants to PR the `git submodule` to match, that's fine by me.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5810)
closed 2017-04-13 by david-german-tri