RobotLocomotion/drake #5793: FindMatlab.cmake does not work for MATLAB R2017a

hongkai-dai, 2017-04-11:
When building with MATLAB R2017a, @RussTedrake and I receive the following error
```
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Matlab: Found unsuitable version "unknown", but required is
  at least "7.7" (found /usr/local/MATLAB/R2017a/extern/include)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/3.7/FindMatlab.cmake:1529 (find_package_handle_standard_args)
  CMakeLists.txt:374 (find_package)

-- Configuring incomplete, errors occurred!
```
I do not see this error with MATLAB R2016a, or any previous versions.

RussTedrake, 2017-04-13:
Note that this only fails when WITH_GUROBI=ON.  drake and the other externals (that I have tried) do not have the same issue.

jamiesnape, 2017-04-13:
Try https://github.com/RobotLocomotion/gurobi/pull/18.

jamiesnape, 2017-04-25:
#5795 is ready for testing and review.

PR #5795 (2017-05-04): Support MATLAB R2016b and R2017a in CMake build
Fixes #5793.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5795)
closed 2017-05-04 by soonho-tri