RobotLocomotion/drake #5799: Valkyrie balancing controller example aborts

rdeits, 2017-04-11:
Running the Val balancing controller aborts with

```
INFO: Running command line: bazel-bin/drake/examples/QPInverseDynamicsForHumanoids/system/valkyrie_balancing_controller_system
abort: failure at drake/examples/QPInverseDynamicsForHumanoids/system/valkyrie_balancing_controller_system.cc:36 in controller_loop(): assertion 'this example is temporarily broken, and will be fixed after #5672 is merged' failed.
```

Looks like maybe that should have been removed as part of #5672 ?

rdeits, 2017-04-11:
Oh, maybe I just misunderstood the message, as in, it will be fixed at some point after #5672 not necessarily as a part of #5672

liangfok, 2017-04-11:
I believe the fix is part of #5787.

rdeits, 2017-04-11:
Got it, thanks.

closed 2017-04-11 by rdeits