RobotLocomotion/drake #5804: yaml_to_obj: error when different lanes have overlapping points

basicNew, 2017-04-12:
Even though it is not a common case, when dealing with arbitrary road geometries two different roads may have an overlapping control point, which causes `yaml_to_obj` to fail. To depict the situation, the following three monolane files are valid:

``` 
# -*- yaml -*-
---
# distances are meters; angles are degrees.
maliput_monolane_builder:
  id: Cross
  lane_bounds: [-2, 2]
  driveable_bounds: [-4, 4]
  position_precision: 0.01
  orientation_precision: 0.5
  points:
    s1_1_1:
      xypoint: [-50.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]
    s1_1_2:
      xypoint: [50.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]
    s2_1_1:
      xypoint: [0.0, 50.0, -90.0]
      zpoint: [0.0, 0, 0, 0]
    s2_1_2:
      xypoint: [0.0, -50.0, -90.0]
      zpoint: [0.0, 0, 0, 0]
  connections:
    s1_1_1-s1_1_2: {start: points.s1_1_1, length: 100.0, explicit_end: points.s1_1_2}
    s2_1_1-s2_1_2: {start: points.s2_1_1, length: 100.0, explicit_end: points.s2_1_2}
  groups: {}
```
[image]

```
# -*- yaml -*-
---
# distances are meters; angles are degrees.
maliput_monolane_builder:
  id: Cross
  lane_bounds: [-2, 2]
  driveable_bounds: [-4, 4]
  position_precision: 0.01
  orientation_precision: 0.5
  points:
    s1_1_1:
      xypoint: [-50.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]
    s1_1_2:
      xypoint: [0.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]
    s1_1_3:
      xypoint: [50.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]    
    s2_1_1:
      xypoint: [0.0, 50.0, -90.0]
      zpoint: [0.0, 0, 0, 0]
    s2_1_2:
      xypoint: [0.0, -50.0, -90.0]
      zpoint: [0.0, 0, 0, 0]
  connections:
    s1_1_1-s1_1_2: {start: points.s1_1_1, length: 50.0, explicit_end: points.s1_1_2}
    s1_1_2-s1_1_3: {start: points.s1_1_2, length: 50.0, explicit_end: points.s1_1_3}
    s2_1_1-s2_1_2: {start: points.s2_1_1, length: 100.0, explicit_end: points.s2_1_2}
  groups: {}
```

[image]

```
# -*- yaml -*-
---
# distances are meters; angles are degrees.
maliput_monolane_builder:
  id: Cross
  lane_bounds: [-2, 2]
  driveable_bounds: [-4, 4]
  position_precision: 0.01
  orientation_precision: 0.5
  points:
    s1_1_1:
      xypoint: [-50.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]
    s1_1_2:
      xypoint: [50.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]
    s2_1_1:
      xypoint: [0.0, 50.0, -90.0]
      zpoint: [0.0, 0, 0, 0]
    s2_1_2:
      xypoint: [0.0, 0.0, -90.0]
      zpoint: [0.0, 0, 0, 0]
    s2_1_3:
      xypoint: [0.0, -50.0, -90.0]
      zpoint: [0.0, 0, 0, 0]  
  connections:
    s1_1_1-s1_1_2: {start: points.s1_1_1, length: 100.0, explicit_end: points.s1_1_2}
    s2_1_1-s2_1_2: {start: points.s2_1_1, length: 50.0, explicit_end: points.s2_1_2}
    s2_1_2-s2_1_3: {start: points.s2_1_2, length: 50.0, explicit_end: points.s2_1_3}
  groups: {}
```

[image]

However, the following monolane definition fails with:
```
[2017-04-12 11:35:25.438] [console] [info] Loading road geometry.
[2017-04-12 11:35:25.441] [console] [error] Lane l:s1_1_2-s1_1_3[start] orientation is off by 2.22144.
[2017-04-12 11:35:25.441] [console] [error] Lane l:s1_1_1-s1_1_2[end] orientation is off by 2.22144.
abort: failure at drake/automotive/maliput/monolane/builder.cc:308 in Build(): assertion 'failures.size() == 0' failed.
Aborted (core dumped)
```

```
# -*- yaml -*-
---
# distances are meters; angles are degrees.
maliput_monolane_builder:
  id: Cross
  lane_bounds: [-2, 2]
  driveable_bounds: [-4, 4]
  position_precision: 0.01
  orientation_precision: 0.5
  points:
    s1_1_1:
      xypoint: [-50.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]
    s1_1_2:
      xypoint: [0.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]
    s1_1_3:
      xypoint: [50.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]    
    s2_1_1:
      xypoint: [0.0, 50.0, -90.0]
      zpoint: [0.0, 0, 0, 0]
    s2_1_2:
      xypoint: [0.0, 0.0, -90.0]
      zpoint: [0.0, 0, 0, 0]
    s2_1_3:
      xypoint: [0.0, -50.0, -90.0]
      zpoint: [0.0, 0, 0, 0]  
  connections:
    s1_1_1-s1_1_2: {start: points.s1_1_1, length: 50.0, explicit_end: points.s1_1_2}
    s1_1_2-s1_1_3: {start: points.s1_1_2, length: 50.0, explicit_end: points.s1_1_3}
    s2_1_1-s2_1_2: {start: points.s2_1_1, length: 50.0, explicit_end: points.s2_1_2}
    s2_1_2-s2_1_3: {start: points.s2_1_2, length: 50.0, explicit_end: points.s2_1_3}
  groups: {}
```

If I do a slight change in one of the points then it works again:

```
# -*- yaml -*-
---
# distances are meters; angles are degrees.
maliput_monolane_builder:
  id: Cross
  lane_bounds: [-2, 2]
  driveable_bounds: [-4, 4]
  position_precision: 0.01
  orientation_precision: 0.5
  points:
    s1_1_1:
      xypoint: [-50.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]
    s1_1_2:
      xypoint: [0.0, 0.1, 0.0]
      zpoint: [0.0, 0, 0, 0]
    s1_1_3:
      xypoint: [50.0, 0.0, 0.0]
      zpoint: [0.0, 0, 0, 0]    
    s2_1_1:
      xypoint: [0.0, 50.0, -90.0]
      zpoint: [0.0, 0, 0, 0]
    s2_1_2:
      xypoint: [0.0, 0.0, -90.0]
      zpoint: [0.0, 0, 0, 0]
    s2_1_3:
      xypoint: [0.0, -50.0, -90.0]
      zpoint: [0.0, 0, 0, 0]  
  connections:
    s1_1_1-s1_1_2: {start: points.s1_1_1, length: 50.1, explicit_end: points.s1_1_2}
    s1_1_2-s1_1_3: {start: points.s1_1_2, length: 49.9, explicit_end: points.s1_1_3}
    s2_1_1-s2_1_2: {start: points.s2_1_1, length: 50.0, explicit_end: points.s2_1_2}
    s2_1_2-s2_1_3: {start: points.s2_1_2, length: 50.0, explicit_end: points.s2_1_3}
  groups: {}
```
[image]

So I was wondering if this is the intended behavior or maybe a bug in the tool.

basicNew, 2017-04-12:
FYI @agalbachicar

basicNew, 2017-04-19:
Probably related to https://bitbucket.org/ekumen/terminus-simulation/issues/119

basicNew, 2017-04-19:
@maddog-tri do you have any insights on this?

maddog-tri, 2017-04-20:
I do!  Connections are made between endpoints, and endpoints are matched by `(x,y,z)` position.  The name strings in a yaml file don't have any significance beyond looking up an endpoint's actual position/etc parameters.

In the "broken" example above, points `s1_1_2` and `s2_1_2` have the same `(x,y,z)` coordinates, so all the Lanes which terminate at those endpoints are linked together at a single BranchPoint.  Finally, the invariants-checker sees that BranchPoint and says "Hey, the orientation of a Lane at this BranchPoint is way off from that of the other Lanes." (like, 90 degrees off).

basicNew, 2017-04-20:
Thanks for the reply @maddog-tri Just to make sure I understand, is this correct?

- In monolane, the only way to describe a road geometry is to do so by points connected between line segments or arcs. 
- From what you are saying there is no logical concept of a lane as a composition of points+connections; each connection in the yaml file is treated like a lane, and both start and end points are treated as lane endpoints.
- Hence, in the example I sent, while logically I intended to have two lanes (the s_1_* and s_2_*), maliput sees this as four lanes.
- Since there are two endpoints that overlap, it assumes there is a branch point and hence the failure.

Also, is this restriction at the monolane level or something enforced by Maliput design?

maddog-tri, 2017-04-20:
>In monolane, the only way to describe a road geometry is to do so by points connected between line segments or arcs.

Correct.  The geometry of a `monolane::Lane` is either single line segment or single arc segment.

>From what you are saying there is no logical concept of a lane as a composition of points+connections; each connection in the yaml file is treated like a lane, and both start and end points are treated as lane endpoints.

Correct.  There is no composition of line-segments/arcs within a single `monolane::Lane`, and each yaml connection becomes a single Lane.

>Hence, in the example I sent, while logically I intended to have two lanes (the s_1_* and s_2_*), maliput sees this as four lanes.

Technically speaking:  `monolane::Builder` is assembling four Lanes.

>Since there are two endpoints that overlap, it assumes there is a branch point and hence the failure.

Technically speaking:  there are four Connections with four overlapping Endpoints, which are being assembled into four Lanes joined at a single BranchPoint.
 
>Also, is this restriction at the monolane level or something enforced by Maliput design?

This is almost all happening in `monolane` --- the only thing that `maliput::api` cares about is enforcing G1-continuity at the BranchPoints.

Is there a reason why `s_1_*` and `s_2_*` are split into two connections each?  Geometrically-speaking, each pair of collinear connections could be replaced by a single connection, eliminating the common central endpoint altogether.

basicNew, 2017-04-20:
> Is there a reason why s_1_* and s_2_* are split into two connections each? Geometrically-speaking, each pair of collinear connections could be replaced by a single connection, eliminating the common central endpoint altogether.

Completely agree that for this case there is no need to have that point, just wanted to depict the issue (we actually have a step where we merge two contiguous line segments if they are collinear). I think the key here is that points are used to describe both geometry and logical connections; in some cases you just get overlapping points because they are describing a road geometry, even though they don't describe a connection. As a very rough example consider the following, where the actual connections for the car between the blue and pink lanes are made by the red arcs. I this case, the end of the arc used to describe the blue lane happens to be the same as the start of an arc describing the pink geometry, causing a similar error as the one described earlier.

[image]

As I mentioned at the start of the issue, this is definitely not a usual situation, but sometimes when deriving monolane files form arbitrary sources it can happen.

jwnimmer-tri, 2017-05-05:
All issues must have owners; feel free to reassign elsewhere as needed (or close), etc.

basicNew, 2017-05-10:
@maddog-tri I understand that this a core part on how monolane is defined and specifying logical lanes connections independently of the geometry seem to be out of its scope. We are currently moving the waypoints by a small delta in case of overlapping when generating monolane files, so if you agree I'll close this one.

jwnimmer-tri, 2019-06-19:
Per #11603, the automotive folder is deprecated, so I'll close this issue as "won't fix"; @basicNew if this issue is still important to you, please re-open it in a different issue tracker.

\CC @liangfok FYI

closed 2019-06-19 by jwnimmer-tri