RobotLocomotion/drake #5805: adding named accessors in AbstractValues / DiscreteValues?

siyuanfeng-tri, 2017-04-12:
the application people are adding more states now, it would be a nice feature to have a named mapping from string to state or something like that inside these. 

are there any objections from the architects? @david-german-tri @sherm1 @jwnimmer-tri 
cc @sammy-tri @naveenoid

david-german-tri, 2017-04-12:
Why not just express this on the System, as we do for ports?  `get_foo_parameter(const Context&)`

jwnimmer-tri, 2017-04-12:
`std::string` use would fail at runtime, instead of compile-time.  Using functions, methods, enumerated constants, or other C++ sugar would be way better than `std::string`.

jwnimmer-tri, 2017-04-12:
I saw there was discussion on a PR about clumsiness around this.  Maybe just cite the code in question, so we have something specific to iterate on fixing?

siyuanfeng-tri, 2017-04-12:
Why not just express this on the System, as we do for ports? get_foo_state_variable(const Context&)
would be ok. but at the bottom accessor, it still rely on some hard coded index number. i think is fine as long as all the indices are kept in the same place.Now suppose we have some derived system that needs to have more abstract states that the base system doesn't have, we start to have indices spread around all of the place. and this causes some concerns
https://reviewable.io/reviews/robotlocomotion/drake/5753#-KhXbk88kwNxJBMkosK2 action_primitive_cc is what triggered this discussion. 

i have tried my best to do what david suggest.  aside from what i said above, the other objection is user needs to type more... 
i see a good argument for failing at compile-time though.

jwnimmer-tri, 2017-04-12:
I will put this on my plate, and look into some proposals ~next week.

siyuanfeng-tri, 2017-04-12:
i am also happy to implement whatever we come up with. 
another option is to replicate the io port setup, have a declare that returns the actual index. i think @jwnimmer-tri mentioned that last friday meeting.

jwnimmer-tri, 2017-04-12:
That would be a great place to start.  I thought we'd already done that.

siyuanfeng-tri, 2017-04-12:
not for abstract mumble? 
it also would change the semantics of DeclareDiscreteState to be more like "AddDiscreteState".

siyuanfeng-tri, 2017-04-12:
will include a draft here at some point. 
the basic idea is to replicate the declare input / ouput use case.

siyuanfeng-tri, 2017-04-13:
@jwnimmer-tri https://github.com/siyuanfeng-tri/drake/commit/416fda45407cc6e9f54767a54d80c4e8aa4e1a0f
something like this?

jwnimmer-tri, 2017-04-13:
API looks fine.  I would have thought the implementation should use `detail::ModelValues` though, instead of a `vector<unique_ptr<>>`.

siyuanfeng-tri, 2017-04-13:
oh... i don't even know about this thing.. sure i will take a look

siyuanfeng-tri, 2017-04-13:
ok. fine. it looks just like a wrapper over the vector of uni ptrs.

siyuanfeng-tri, 2017-04-13:
https://github.com/RobotLocomotion/drake/pull/5819

closed 2017-04-17 by siyuanfeng-tri