RobotLocomotion/drake #5812: DepthSensor Roadmap For Supporting Both Kinematic and Dynamic Simulations

liangfok, 2017-04-13:
# Problem Definition

`DepthSensor` currently only works with `RigidBodyTree`. This is insufficient because (1) we also want to use it with kinematics-only simulations, and (2) `RigidBodyTree` is about to be deprecated in favor of `MultibodyTree`.

# Current Architecture

The current architecture is shown below in Figure 1. The generalized joint configuration (`Xc`) output of a `RigidBodyPlant` is an input into `DepthSensor`. The `RigidBodyPlant` owns a `RigidBodyTree`, which is referenced by the `DepthSensor`. The `DepthSensor` uses the `RigidBodyTree` to perform collision ray casts to mimic the depth measurements obtained by a simulated sensor.

[image]

_**Figure 1:** The original architecture. `RigidBodyPlant` owns an instance of `RigidBodyTree`. `DepthSensor` has a reference to that instance and uses it to invoke `collisionRayCast()` to define output._

# Proposed Intermediate Architecture Using PoseAggregator

The proposed architecture is shown below in Figure 2. Instead of receiving `Xc`, the `DepthSensor` receives a `PoseBundle` from a `PoseAggregator`. The `PoseBundle` contains the poses of every model instance in the simulation. Each model instance can have any number of geometry elements. The `DepthSensor` still uses a `RigidBodyTree`, but the tree is owned by the `DepthSensor` and is specially configured so that every model instance is actually a collection of welded `RigidBody` objects and each model instance is connected to the world via a quaternion floating joint. By using a quaternion floating joint, `DepthSensor` can easily take the poses within the provided `PoseBundle`, internally convert them into `Xc`, and use this with its `RigidBodyTree` to make the necessary `collisionRaycast()` calls.

Hybrid dynamic + kinematic simulations are supported since PoseAggregator can aggregate the poses from both dynamics-based systems (e.g., RigidBodyPlant, MultibodyPlant) and kinematics-based systems (e.g., SimpleCar, TrajectoryCar, BicycleCar, MaliputRailcar, etc.)

[image]

_**Figure 2:** Proposed intermediate architecture. `DepthSensor` keeps its own copy of a `RigidBodyTree` instance. The tree serves as a mechanism for up performing ray cast queries. The bodies' poses are updated from the input poses prior to the queries._

# Proposed Long Term Architecture Using GeometrySystem

When `GeometryWorld` and `GeometrySystem` exist, the `DepthSensor` won't have to maintain any internal state or know any kinematics. Kinematics data feeds into `GeometrySystem` and `DepthSensor` connects a `GeometryQuery`-abstract-valued input port to `GeometrySystem`'s corresponding output port. All ray-casting queries are performed on the `GeometryQuery` instance obtained through `DepthSensor`'s input port.

[image]

_**Figure 3:** `GeometrySystem`-based architecture. `DepthSensor` no longer knows anything about kinematics. An abstract-valued output of `GeometrySystem` provides an instance of context-dependent `GeometryQuery`. `DepthSensor` performs ray-casting queries on this instance._

liangfok, 2017-04-13:
/cc: @SeanCurtis-TRI since this involves `GeometryWorld`. Also, thanks for generating the figures!

SeanCurtis-TRI, 2017-04-13:
Looks good.

liangfok, 2017-04-13:
I just updated the OP to relax the requirements of the `RigidBodyTree` within the `DepthSensor`. I no longer think it's necessary to weld all of the bodies belonging to a model instance, nor do I think it's necessary to restrict the floating joint to be of type quaternion.

sherm1, 2020-05-11:
@SeanCurtis-TRI should this be closed now?

SeanCurtis-TRI, 2020-05-12:
We *could*. The thing that is in attic is not quite the same as the depth image in `RgbdSensor`. This actually did structured ray casting. So, we can close it and if we end up with depth needs that can't be satisfied with a rasterized depth image and, in fact, need to support the ray casting queries, then we can re-open it.

closed 2020-05-12 by SeanCurtis-TRI