RobotLocomotion/drake #5813: Context::FixInputPort() or System::FixInputPort()?

amcastro-tri, 2017-04-13:
I find it strange (and this is a personal opinion) having to do `context->FixInputPort()` to wire up a constant source into a system. I think it'd lead to a more homogeneous API if we had something more like a `System::FixInputPort()` (not necessarily with this syntax but mainly avoiding to ask the question to Context) that would return a properly wired up `Diagram`.
The reason why I am bringing this up is because in a recent [PR review](https://reviewable.io/reviews/robotlocomotion/drake/5575#-KhN_74eVDwW_ZWEI-vJ) we found we have to take care of input ports separately when integrating a system in time. I find that a bit error prone. In my opinion `Simulator` should only have to deal with full `Diagram` models that have no inputs, they are isolated from the universe and we can only "look" at them (so no wired outputs would be allowed).

Thoughts?

david-german-tri, 2017-04-13:
The thread you linked from #5575 looks like a non-problem to me.  The author's code was correct before you commented on it.  Using existing sugar, it converts a System from `double` to `AutoDiffXd`, generates a new `Context`, and copies the `double` values from the old `Context` to the new `Context` `.value()` parts.  Local linearization about a point, in four lines of code!

If you have an idea to do better, I think it would be easier to discuss after you've written a prototype.  If you're introducing a new requirement that Systems under Simulator have zero inputs, please also add an assert for that and migrate the test cases that break.

amcastro-tri, 2017-04-13:
> The author's code was correct before you commented on it. 

Yes, I am not marking a defect, that code just triggered a questions I had for a while now, that's it. I just decided to see what others think. The answer might be that this is exactly what we want and I'd be ok.

I'll quote @edrumwri here from the same thread:
> Short answer: the DoCalcTimeDerivatives() method will throw an exception otherwise (for systems that require inputs), as it expects inputs to be present.

I just wonder how much mental effort it took @edrumwri to arrive to the conclusion that he should call `FixInputPortsFrom()` from that exception.

Again, this is just a question, not a proposal and I am not even sure I identified a real problem. That is whey I labeled "question".

sherm1, 2017-04-13:
From slack discussion on #system_framework: I think this is OK but would be perfect if the input port value transmogrification could be done in the same Context method call as time, parameters, and state rather than in a separate System method.

amcastro-tri, 2017-04-13:
That'd be super nice I think but it seems to fall in the category of "syntactic sugar". Maybe we could add that feature if we need it somewhere else.

closed 2017-04-13 by amcastro-tri

sherm1, 2017-04-13:
> That'd be super nice I think but it seems to fall in the category of "syntactic sugar".

Almost, but not quite because it implies that the input port transmogrification can be done entirely by the context without the System's help. I *think* that is possible, but it is not just a syntactic change.

amcastro-tri, 2017-04-13:
I am copying here the conversation we had in Slack this morning about this. Thank you everybody for your input.

jwnimmer-tri [9:10 AM] 
I find it a strange and confusing request.  If you want to facilitate testing, write better test sugar or fixtures.  I don't yet see a reason why System itself must be the receiver of this sugar.
 3 replies Last reply today at 9:51 AM View thread

jwnimmer-tri [9:15 AM] 
To be clear, ripping the Context out of a System is a weird programming model that takes time to understand, but it's tough for _all_ parts of the API, so I don't see why `FixInputPort` should receive special treatment.

alejandro [9:36 AM] 
The reason I asked is that I thought that integrators should only see a function `xdot = f(t, x) ` to integrate, and therefore the system is isolated from the universe. However, they see a function `xdot = f(t, x, u)` and therefore they also have to take care of the inputs. So, doing `CalcTimeDerivatives()` might give you the wrong answer if you forgot to update the inputs, something a bit difficult to track for complex integrators.

[9:41] 
maybe not important since I guess we guarantee those `u` are constant?

david-german-tri [9:41 AM] 
> something a bit difficult to track for complex integrators
Was there more trouble than the thread you linked?  Because that was one line, which the author apparently had right before you commented on it.

[9:43] 
> maybe not important since I guess we guarantee those `u` are constant?
Just to clarify - the Simulator guarantees that, not the Framework. (For instance, @siyuan.feng operates Systems with externally clocked inputs.)

jwnimmer-tri [9:59 AM] 
Yeah, so I went and read the code from the PR.  I agree with David.  In no world should `System::ToAutoDiffXd` snarf things from the double Context and Fix them into the new Context.  If we want to wrap those four lines into a standalone helper function, that might be worthwhile.

alejandro [10:03 AM] 
> Just to clarify - the Simulator guarantees that, not the Framework.
Could you further clarify that point?

[10:04] 
also, sorry, I also replied your first question in the issue I created @david-german-tri

david-german-tri [10:05 AM] 
You can perfectly well change the inputs to a System with custom code - @siyuan.feng does this to clock in inputs from external sources, I think, and the LuenbergerObserver does it for the wrapped System

[10:06] 
However, there's no hook in the Simulator that would synchronize external inputs into the flow of simulation time, so the spirit of your point is correct :slightly_smiling_face:  That's all I meant. (edited)

jwnimmer-tri [10:08 AM] 
If your `f` (time derivatives) is input-dependent, then your integrator has to be employed in a way where the inputs are valid for the time its asking about.  That's the Simulator / Integrator's problem, and is unavoidable for Systems where the input affects the time derivatives (as it often will).  I don't see why that's a complaint; its intrinsic to the problem domain.

jwnimmer-tri [10:18 AM] 
Ah, I see.  The relevant reminder here is not thinking about within a Diagram.  It's literally just a System that you hand to the Simulator.  If it has unbound inputs, the caller will Fix them first, and then the Simulator when it transmogrifies the Context<double> has to also Fix the Context<AutoDiff> to match. (edited)

[10:18] 
Alejandro is proposing that the Simulator can only operate on Systems with no inputs, so you have to cascade a ConstantValueSource into your System before handing to the Simulator, rather than Fix on the Context you give to the simulator.

[10:19] 
As David said on the issue, if you want to propose that fine, but put some code on the table and see if you like it better.

alejandro [10:26 AM] 
>Alejandro is proposing that the Simulator can only operate on Systems with no inputs, so you have to cascade a ConstantValueSource into your System before handing to the Simulator

Yes, exactly right, that was what I was suggesting. I don't really know if it solves anything at this point but I decide to share this thought with you guys. Thank you for the input. Let's give the PA folks time to get to the office and have their coffee to see what they think

sherm1 [10:55 AM] 
Coffee ingested ...
 

sherm1 [11:11 AM] 
A Simulator can only operate on a System that has all its inputs resolved, except for `time` which is the only input the Simulator can generate by itself. One way to resolve the inputs is not to have any, as @alejandro suggested. Others would be to set inputs to default values or to values supplied in the Context. The ability to store input values in the Context is in fact necessary if you hope to be able to capture a trajectory by recording a series of Context values that can be played back or analyzed later. @david-german-tri and I figured that out very early in System 2 design discussions last year.

[11:14] 
So philosophically I think having the Simulator act on Systems with unconnected-but-resolved input ports is fine. If there is anything odd about the code under discussion it is just that the transmogrification of the input port values requires a System method rather than just a Context method like the state & parameter transmogrification does. If state, parameters, time, and input values could all be transmogrified in one Context method it would be perfect.

[11:15] 
cc @drum

alejandro [11:26 AM] 
humm.. ok. All of the comments above make sense to me, clearly not a real problem here. This seems to only bother me so clearly not a real issue and I'm just bikeshedding here. I'll go ahead and close #5813. Thanks for the useful discussion!