RobotLocomotion/drake #5827: Allow naming of Bindings and Constraint Rows for ease of use

EricCousineau-TRI, 2017-04-13:
This is meant to address Hongkai's [comment](https://github.com/RobotLocomotion/drake/issues/3130#issuecomment-248046817) in #3130, that constraints should have names to provide human-friendly context.

## Quick Overview:

User can specify and retrieve (a) a binding name, and (b) a binding's row name (by specifying a constraint row name).

Examples:

*   `linear_constraint208/row15`
*   `torque_bounds:lower/right_knee`
*   `knot5:dynamics_constraint/row15`

Informal Example:

    std::tie<VectorXDecisionVariable, Binding<BoundingBoxConstraint>> AddTorqueBounds(prog, robot, lower, upper, ...) {
        ...
        binding = prog.AddBoundingBoxConstraint(lower, upper, tau, actuator_names); // -> bounding_box208/lower:right_knee, ...
            OR
         = prog.AddConstraint(make_shared<BoundingBoxConstraint>(lower, upper), tau, "torque_bounds") // -> torque_bounds/lower:right_knee, ...
    }

## Schedule

As this is a nice-to-have aesthetic, I will implement this after initial implementations are implemented for (a) evaluating solution violation #4733 and (b) inactive constraints are returned #3130, and possibly before/after (c) active optimal constraints are returned #5654, and definitely before (d) being able to serialize/deserialized `MathematicalProgram`'s.

## Functionality

* Provide the ability to optionally specify unique constraint row names
    * Row names will default to `row${row_index}` (e.g., `row15`)
    * Could be specified as `lower:right_knee`
* Provide the ability to optionally specify unique binding prefixes:
    * By default, there will be a ID counter (the ID will not correspond to the binding's index in the respective container), and will default to `${constraint_type}${id}` (e.g., `linear_inequality208`)
    * Could be specified as `torque_bounds`
* Naming convetions
    * Names should 
    * Names will never be empty.
    * Delimiters between bindings and constraints will be `/`
        * Internal delimiters should be `:`
        * As an initial pass, no checks will be introduced to validate identifiers (e.g., `[a-zA-Z0-9_:]`).
            * Concerns: This permits divergence as people use it...
            * *Alternative*: Do enforce that no top-level delimiters are used, and use an alternative, such as `/` for top-level delimiters, and `:` for lower-level, e.g. `top:sub_top/lower:sub_lower`
    * Indices will be appended directly `${name}${index}` and will be 1-based to maintain consistency with decision variables (e.g., `linear_inequality1`).
        * *Alternatives*: `linear_inequality(1)`, `linear_inequality[0]`

## Implementation Plan

* New Methods and Example Returns:
    * `Constraint::GetRowName(int row_index) const` -> `"row15"`
        * `protected: Constraint::SetRowNames(const vector<string>& names)`
    * `Binding::GetName() const` -> `"linear_inequality208"`
    * `Binding::GetRowFullName(int row_index) const` -> `"linear_inequality208/row15"`
    * `MathematicalProgram::GetNextDefaultBindingId()` -> `208`
        * `MathematicalProgram::GetNextDefaultBindingName(Constraint*)` -> `"linear_inequality208"`
    * Public Helpers:
        * `prefix_string_list(const string& prefix, vector<string>)` -> `{"lower:right_knee", "lower:left_knee", ...}`
* Updates for Existing Methods:
    * `Constraint::Constraint(..., +const vector<string>& names = {})`
        * `LinearConstraint::LinearConstraint(..., +const vector<string>& names = {}) : Constraint(...) : names(names)`
    * `Binding<C>::Binding(..., +const string& name = "")`
    * `Binding<C>::Binding(const Binding<U>&, +const string& default_name)`
    * `MathematicalProgram::Add*Constraint(..., const string& name = "")`
* Bottom-Up / Encapsulation Implementation Issues (for now, going with Option 2 for simplicity)
    * *Issue*: A unique default name cannot be determine if a `Binding` is constructed external to the `MathematicalProgram`.
        * If a default name is used, there will be a discrepancy between a user-constructed `Binding` and the returned `unique_ptr<Binding...>`  name when a user calls `MathematicalProgram::AddConstraint(Binding...)`.
    * Solution Options (will go with Immutable):
        1. Immutable:
            * Because `Binding`s are copied into a `MathematicalProgram`, the name may remain in the constructor.
            * For a copy constructor, a `default_name` can be supplied that will only be used if the rvalue `Binding` has an unset name
        1. Mutable:
            * Add optional `set_name()`, which may only be called if the name is empty.
            * This does not prevent the original issue. There is no mechanism to provide context without violating bottom-up construction.

* Uniqueness will be enforced at each level, defaulting to an error if a name is duplicated at the given scope
    * Unittests will ensure that there are no name collisions
    * Regardless of constraint type, `Binding`s must have unique names; e.g., `prog.linear_constraints_[i]` and `prog.quadratic_constraints_[i]` should not share the same name.

## Precedents:

* `drake/matlab/solvers/NonlinearProgram.m`
    * Functionality: `[infeasible_constraint_name] = ineasibleConstraint(obj,x)`
        * Returns each names constraint, which corresponds to a row within a given `Constraint` MATLAB class
        * This implementation will provide similar granularity, but adds hierarchy to handle the same constraint being used in other contexts
* `drake::solvers::MathematicalProgram::NewVariables_impl`
    * Implementation: TTBOMK, no identifier format checking or uniqueness is enforced. This can be added if necessary, and this issue can be changed.

## TODO

* Review some more precedents
* Find and expand on concrete usages

## Caveats

Discussed with @soonho-tri and @hongkai-dai:

* If the constraint is a vector of `symbolic::Formula`s, the ordering may be permuted and identical constraints may be eliminated. Soonho will provide the resulting mapping (original indices `1:n` -> final indices) that can then be applied to the supplied / default row names.
* For dynamically adding / removing constraints, the original indexing would be invalidated (if the default name contains the ), and may produce collision.
     * This merits the usage of a counter internal to the `MathematicalProgram`
        * *Potential Issue*: Integer overflow if constantly removing / adding constraints?

\cc @hongkai-dai, @soonho-tri, @RussTedrake, @siyuanfeng-tri, @rdeits

EricCousineau-TRI, 2017-04-19:
Will review solvers to see if it may be convenient to up this priority if association by name can be leveraged in mapping.

[Incomplete] Solver List, and name association abilities when adding constraints:

* Gurobi - Can specify via `GRBaddconstr(..., const char* name)`

Will review the remaining solvers to see what the most convenient naming convention is.

Additional design case:

* Specific solvers may add additional forms of a given constraint (e.g., LCP). It is necessary to have solver-specific mapping, and this may be provided by solver-specific names in addition to what the existing `Binding` has.

jwnimmer-tri, 2019-03-26:
@EricCousineau-TRI has this been a pain point recently?  If not, maybe we should close this?

EricCousineau-TRI, 2019-03-26:
Nope. And this def. requires more API churn than worth at the moment. I'm up for re-opening if someone else encounters this pain point.

closed 2019-03-26 by EricCousineau-TRI