RobotLocomotion/drake #5850: bazel builds ipopt too often

RussTedrake, 2017-04-16:
In my current workflow, I build bazel BOTH from clion and from the command line.  Every time I switch between the two, I see
```
Executing genrule @ipopt//:build_with_autotools
```
which takes about 130s on my virtual machine.  To be clear, I believe this is triggered only when I switch, e.g.

build from clion  - ipopt builds
build from clion
build from clion
build from shell - ipopt builds
build from shell
build from clion - ipopt builds
...

david-german-tri, 2017-04-17:
Speculation: some flag or environment variable is different, and Bazel isn't sure whether that affects the `ipopt` build, so it rebuilds to be safe.  I'll try to reproduce.

As a workaround if you don't actually need IPOPT, you can squelch it with `--define=NO_IPOPT=ON` in your `~/.bazelrc`.

RussTedrake, 2017-04-17:
Thanks.  That's my guess as well.  This is a pretty fresh ubuntu machine with clion 2016.3 (number is from memory, but it's before the 2017 rev)

david-german-tri, 2017-04-20:
I've root-caused this.  Bazel 0.4.5 introduced the (really useful) command-line flag [--action_env](https://bazel.build/designs/2016/06/21/environment.html).  If you pass `--action_env FOO`, Bazel will rebuild targets that are sensitive to the external shell environment (such as the IPOPT genrule) whenever `FOO` changes.  By default in the systemwide `/etc/bazel.bazelrc`, you have the following settings.

```
build --action_env=PATH
build --action_env=LD_LIBRARY_PATH
build --action_env=TMPDIR
build --test_env=PATH
build --test_env=LD_LIBRARY_PATH
```

Seems completely reasonable, right?  You definitely want to rebuild IPOPT if the path changes.  Well, as it turns out, when CLion on Linux launches build subprocesses, it gives them the same `PATH` as the shell in which it was launched, but it prepends `./bin:` to the `LD_LIBRARY_PATH`.  Why, I have no idea.

As a stupid workaround that proves the hypothesis, you can `export LD_LIBRARY_PATH=./bin:` in your console, build `ipopt_solver` from CLion, and then build it from the console, yielding a cache hit.

I'm going to look for mitigations within CLion.  There might be an option somewhere in the settings, or this might be a CLion bug, or it might (unlikely) be a Bazel-for-CLion plugin bug.

david-german-tri, 2017-04-21:
I've determined this is not a plugin behavior: the mangled LD_LIBRARY_PATH shows up in CLion CMake projects too.  I can't find a knob anywhere in the CLion UI to make it stop, so I've opened https://youtrack.jetbrains.com/issue/CPP-9585 - please upvote!   Nothing else we can do, so closing as infeasible.

closed 2017-04-21 by david-german-tri

jwnimmer-tri, 2017-04-21:
Could we work around this by adding `build --action_env=LD_LIBRARY_PATH=` to our `tools/bazel.rc`?  In other words, always null out the library path, no matter how Bazel was invoked?

david-german-tri, 2017-04-21:
> Could we work around this by adding build --action_env=LD_LIBRARY_PATH= to our tools/bazel.rc? In other words, always null out the library path, no matter how Bazel was invoked?

Good idea - that works!  I'm leery of checking it in, but it's a nice answer for `~/.bazelrc` at a minimum.