RobotLocomotion/drake #5856: Serialize Context (using protobuf)

RussTedrake, 2017-04-17:
Potential use case:  we find a falsification scenario (e.g. from running many simulations) that causes our controller to fail.  We'd like to write that "scenario" to disk and add (once the issue is resolved) add it to the test suite as a regression test.  

We've architected this carefully already -- everything we need to know (assuming we have the identical system in memory) should be in the Context.  We just need to serialize it.

jwnimmer-tri, 2020-02-06:
As of #12689 we are ditching protobuf, and we should probably not pick it up again.  I've edited the title to make protobuf a placeholder instead of a definitive technology choice.

jwnimmer-tri, 2021-11-11:
We have YAML serialization available in Drake now.  If all of the relevant abstract values provided Serialize function, I think this is cake.