RobotLocomotion/drake #5857: Print Context information e.g. implement Context::operator<<(stream..)

RussTedrake, 2017-04-17:
Writing optimizations over big complicated diagrams is currently very difficult, because I don't have a good way to understand the resulting composite input, state, and parameter vectors.  A simple routine that serializes a textual version of this information to the display could work wonders.  It might be a blocker for me to get much farther on my automotive optimization work.

When systems and or basic vector elements are given explicit names, those should be used in the printout.

liangfok, 2017-04-18:
To make the printout pretty, I'm thinking about using JSON for modern C++: https://nlohmann.github.io/json/. It is MIT-licensed.

liangfok, 2017-04-18:
I have a WIP branch here: https://github.com/liangfok/drake/tree/feature/context_to_string

RussTedrake, 2017-04-18:
Great!  I think if we're going to serialize to protobuf, then using protobuf here instead of json might make sense, too?

liangfok, 2017-04-18:
Good point. I briefly considered using Protocol Buffers but then realized that I would likely need to define a `.proto` message for each class being serialized. This could get complicated when attempting to serialize abstract values.  After talking F2F with @david-german-tri, I decided to move away from JSON since the immediate goal is not serialization but rather debugging output.

jwnimmer-tri, 2017-04-20:
You can very well use `proto` in a general way, without a message for each class being serialized.  In fact, #5856 (correctly) presumes that we'll be able to do this.  Doing a single thing to solve both that issue and this one seems like the best approach?

liangfok, 2017-04-20:
Alright, I think this will require an initial design review. I'll start working on one ASAP.

liangfok, 2017-04-21:
From F2F conversations yesterday, I now believe there are two approaches to addressing this issue:

1. Within the context of Drake's `AutomotiveSimulator`.
2. Within Drake's system framework as a generally-supported feature.

Option 1 can be done quickly and is the approach taken by #5910. Option 2 is very tricky since it has less control over both diagram topology and leaf system types. It will likely take many weeks and involve a careful design review phase.

RussTedrake, 2017-04-22:
Let's leave this open until we address #2 -- that is the ask.  #1 might be sufficient for now (will see how far I get this weekend).

liangfok, 2017-04-22:
Alright. I just updated #5910 to include `MaliputRailcar` information since I believe you're using them.

jwnimmer-tri, 2021-11-11:
FYI I expect this to become pretty easy now that we have YAML serialization in `drake/common/yaml`.

RussTedrake, 2022-11-18:
We can already print the context, and serializing to YAML is covered by #5856.  Will close this one.

closed 2022-11-18 by RussTedrake