RobotLocomotion/drake #5862: Gurobi license has expired in CI

david-german-tri, 2017-04-17:
Until the license is restored, I'm going to disable all tests that require Gurobi.

david-german-tri, 2017-04-17:
#5863 revealed the need for #5864.

david-german-tri, 2017-04-17:
#5863 appears likely to resolve the emergency - CI is returning green so far.

Follow-up actions:

- [ ] Lift the current moratorium on merges.  @david-german-tri
- [ ] Install a new Gurobi license in CI.  @RussTedrake to obtain, @BetsyMcPhail to install.
- [ ] Revert #5863.  @BetsyMcPhail after installing.
- [ ] Create a reproducing test case for the probable Bazel bug in #5864, and file it.  @david-german-tri

david-german-tri, 2017-04-17:
Done with my tasks on this issue - remaining items are for @BetsyMcPhail, following up with @RussTedrake as needed.

RussTedrake, 2017-05-27:
Resolved a few weeks back

closed 2017-05-27 by RussTedrake