RobotLocomotion/drake #5869: Consider reimplementing inverseKin / inverseKinTraj interfaces

EricCousineau-TRI, 2017-04-17:
Per Hongkai's and Sammy's prior discussions, the present implementations of `inverseKin` have some drawbacks that could be addressed by refactoring / reimplementation.

Some items:
* #3675 - No access to the `MathematicalProgram`[s] used to solve the problem.
* `infeasible_constraints` are no longer returned by the present implementation.
* `info` goes through semi-redundant mapping: from `inverseKin(...): SolutionResult result = prog.Solve();`, it is mapped to SNOPT-style codes at `inverseKin(...): info[t_index] = GetIKSolverInfo(prog, result);`.
    * If this is for maintain MATLAB backwards-compatibility, this can be addressed within the wrapper.
* `RigidBodyConstraint` uses id-based type erasure which may not be necessary
* _Will add more as they come up..._

\cc @hongkai-dai @sammy-tri @siyuanfeng-tri

RussTedrake, 2017-04-18:
In particular, we've discussed implementing inverseKinTraj as a trajectory optimization (using dircol, etc) on a kinematic-only version of rigid body plant.  @avalenzu did a version of this in matlab, and we liked it a lot.  See, e.g. http://drake.mit.edu/doxygen_matlab/class_kinematic_dirtran.html

jwnimmer-tri, 2017-05-05:
All issues must have owners; feel free to reassign elsewhere as needed (or close), etc.

EricCousineau-TRI, 2018-12-08:
@hongkai-dai @avalenzu I believe y'all are already handling this. Should I close this if we have another tracking issue? If not, can I reassign one of y'all?

avalenzu, 2018-12-10:
AFAIK, we don't have another tracking issue. Feel free to assign it to one of us.

EricCousineau-TRI, 2018-12-10:
K, done.

hongkai-dai, 2018-12-10:
Just saw this. I think I would like to split this issue to two parts
1. Reimplement inverse kinematics to use MathematicalProgram interface and MBP. This is related to #9341.
2. Implement kinematic trajectory optimization. I think this is related to Andres' B-spline approach. Or we can consider other approaches to interpolate the kinematic trajectory.

avalenzu, 2018-12-11:
I'd be on board with that split, @hongkai-dai.

tri-ltyyu, 2018-12-11:
#10188 created for the trajectory optimization.

closed 2019-01-25 by hongkai-dai