RobotLocomotion/drake #5873: Ipopt returns different result between mac-clang-bazel and linux-clang-bazel

EricCousineau-TRI, 2017-04-18:
Ipopt returns a different result when running on Mac vs Linux for a purposely infeasible IK optimization problem (`//drake/multibody:test_ik`)

`mac-clang-bazel-*` - see [full log](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-clang-bazel-nightly-release/119/consoleFull) at `08:44:40`
`linux-clang-bazel-*` - see [full log](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-clang-bazel-nightly-release/lastBuild/consoleFull) at `01:41:37 PASS: //drake/multibody:test_ik`

Possible solutions (per discussions on #buildcop):
* Ensure that status is appropriately mapped
* Use `select` and `config_setting` in Bazel to disable this test on Mac when `snopt` is unavailable

EricCousineau-TRI, 2017-04-18:
Realized that the returned code (3) from `GetIKSolverInfo` is not Ipopt's flag (`Search_Direction_Becomes_Too_Small=3`), but rather `SolutionResult::kIterationLimit`. Will update test to check for either of these flags.

This still doesn't answer the question why Ipopt returns different results between two platforms, but will at least squelch the test error.

EricCousineau-TRI, 2017-04-18:
This error appears to be successfully squelched.

[CI full log](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-clang-bazel-nightly-release/120/console):
`14:22:12 PASS: //drake/multibody:test_ik`

Keeping issue open due to discrepancy.

jwnimmer-tri, 2017-05-05:
All issues must have owners; feel free to reassign elsewhere as needed (or close), etc.

jwnimmer-tri, 2017-09-20:
Note that #7027 (if merged) will add even more possible differences -- the Mac and Ubuntu versions of IPOPT will not be identical, since we will be using the platform default.

EricCousineau-TRI, 2018-12-08:
Closing as I'm assuming we won't be able to fix this anytime in the near future.

closed 2018-12-08 by EricCousineau-TRI