RobotLocomotion/drake #5890: Refactor constraint- and cost-related methods in MathematicalProgram to decouple Constraint and Binding constructors from MathematicalProgram interface

EricCousineau-TRI, 2017-04-19:
## Overview

Refactor constraint- and cost-related methods in `MathematicalProgram` to decouple `Constraint` and `Binding` constructors from `MathematicalProgram` interface.

This could cause usage to change from:

    prog.AddLinearComplementarityConstraint(M_, q_, vd_);

to

    prog.Add(LinearComplementarityConstraint(M_, q_), vd_);

but would help in making the role of `MathematicalProgram` more succinct (a registration container, not a factory).

This is more broadly scoped to address the heart of the issue, and 
avoid fighting the language as in #5885.

This scope is also different from #4559 in that this focuses on `MathematicalProgram`s role as a generalized container for constraints and costs.

\cc @soonho-tri @hongkai-dai @sammy-tri @ggould-tri @siyuanfeng-tri @RussTedrake

## Motivation

* Simplify and decouple interfaces
* No one appears to be externally consuming the direct interface to `MathematicalProgram` yet, so now may be a good (easier) time to change it
* Additional motivations from #5885

## Functionality

* Make the newer interface as succinct as before, maybe with just a few more added symbols
* Avoid attempting to do specializations with parameter packs like the plague
    * Sticky issues with greedy template consumption and the various references types that would normally match a template during normal inference (see #5885 for more details)

## PR Plan

1. Implement Step 1 from #5885
2. Implement Suggested Interface

#### Current Interface

(Copied from #5885)

Current form of functions (return types, arguments, and high-level overview):

1. `Binding<${CONSTRAINT}> AddConstraint(...)`
    1. `ctor_args<Binding<${CONSTRAINT}>>...`
        * Construct the specific `Binding` (if necessary), and add to appropriate container
1. `Binding<${CONSTRAINT}> Add${CONSTRAINT}(...)`
    1. `ctor_args<${CONSTRAINT}>..., ctor_args<Binding<${CONSTRAINT}>>(const shared_ptr<${CONSTRAINT}>&, ...)...`
        * Construct the specific `Constraint` (if necessary)
        * Create the specific `Binding`, and then call `AddConstraint(...)`
1. `Binding<Constraint> AddPolynomialConstraint(...)`
    1. `ctor_args<PolynomialConstraint>..., ctor_args<Binding<Constraint>>(const shared_ptr<Constraint>&, ...)...`
1. `typename std::enable_if<..., Binding<${CONSTRAINT}>::type Add${CONSTRAINT}(...)`
    1. `symbolic_args...`
        * Do fancy symbolic constraint inference magic, and then call one of the above methods

Types:

* `ctor_args<${CLASS}>...` - Constructor arguments for a given `${CLASS}`
    * `ctor_args<${CLASS}>(${ARG}, ...)...` - Constructor arguments for an overload beginning with certain argument types, returning all but the explicitly specified arguments
* `symbolic_args...` - Arguments relating to a 
* `${CONSTRAINT}` - A direct placement of `Constraint` or any sub-class. I see repetitions for the following types (amounts to ~1000 / ~2600 lines in `mathematical_program.h`, ~600 / ~1400 lines in `mathematical_program.cc`):
    * `Constraint`
    * `LinearConstraint`
    * `LinearEqualityConstraint`
    * `QuadraticConstraint`
    * `BoundingBoxConstraint`
    * `LorentzConeConsraint`
    * `RotatedLorentzConeConstraint`
[... 120 lines cut ...]

1. Provide the same set of overloads within `${CONSTRAINT}` / `${COST}` constructors, and then add an explicit function for a symbolic version, `symbolic::${CONSTRAINT}` / `symbolic::${COST}`.
    * Examples:
        * `Binding<Constraint> symbolic::Constraint(Formula...)` - dispatch to the appropriate constraint type, if able, at compile time via expression matching and `std::enable_if` magic, or just add a general nonlinear constraint
        * `Binding<LinearConstraint> symbolic::LinearConstraint(Formula...)` - create the appropriate linear constraint with the appropriate checking
    * Pros: Cleaner implementation-wise
    * Cons: Requires a little more typing
2. Provide only one succinct, non-overloaded constructor for each `${CONSTRAINT}` and `${COST}` class.
    * Delegate the different constructors to functions declared in a `drake::sovlers::common` namespace, that can later be used.
    * Examples:
        * `shared_ptr<Constraint> PolynomialConstraint(...)` - non-symbolic overload
        * `Binding<Constraint> PolynomialConstraint(Formula...)` - symbolic overload
    * Pros: Cleaner conceptually
    * Cons:
        * Mixed return types for the same method name
        * May cause symbolic ambiguity if the constraints are visible in the current scope
            * Care should be taken to explicitly avoid this
3. Provide symbolic overloads in each `${CONSTRAINT}`
    * I do not like this option. This means that the `${CONSTRAINT}` must now have knowledge of its symbolic context, and has to optionally store this context, and then the container must determine at runtime whether it needs more information.

My favor is for Option 2, as this is generally the cleanest.

## Potential Drawbacks

* Refactoring in general

hongkai-dai, 2017-04-19:
What is the return type of
```C++
prog.Add(LinearConstraint(A, b), vars);
```
? It seems to me now we can only return `Binding<Constraint>` type, instead of a `Binding<LinearConstraint>` object? I hope we can still get the specific type of the constraint back, since later we may need to update the constraint.

Currently one use case is
```C++
auto binding = prog.AddLinearConstraint(A, lb, ub, vars);
prog.Solve();
// Now update the constraint, and solve the problem again with the updated constraint
binding.constraint()->UpdateConstraint(new_A, new_lb, new_ub);
prog.Solve();
```
Can we still achieve the same "update and resolve problem" using this new interface?

EricCousineau-TRI, 2017-04-19:
The return-type would be the same as the current API. It could return `constraint_binding<C>`, where:
* `constraint_binding<C> = Binding<C>` in the general case
* `constraint_binding<PolynomialConstraint> = Binding<Constraint>`, to permit the constraint to simplified if need be.
    * NOTE: This will require run-time check when `C = Constraint`, but I believe this would be useful (especially in the general symbolic case) and relatively straightforward.

hongkai-dai, 2017-04-19:
If I understand correctly, we will not be able add the symbolic constraint in the following way any more
```C++
prog.AddConstraint(2 * x + 3 * y <= 1); // prog add a LinearConstraint
```
But instead we need to do 
```C++
prog.Add(LinearConstraint(2 * x + 3 * y <= 1)); // The user has to tell the program, that this is a linear constraint.
```
is this the case?

ggould-tri, 2017-04-19:
Presumably it would be straightforward to write a CreateSymbolicConstraint(...) factory that dispatched to the appropriate subclass based on the expression provided?

hongkai-dai, 2017-04-19:
It shouldn't be hard to write the factory. Sorry I did not realize the factory is the plan. If it is not too much of work, could you put the skeleton of this  API for this factory here?

EricCousineau-TRI, 2017-04-19:
Yup! The factory would be something to the effect of:

    // Numeric overload(s)
    shared_ptr<LinearConstraint> CreateLinearConstraint(const Matrix& A, const Vector& b) {
        return make_shared(new LinearConstraint(A, b));
    }
    // Symbolic overload(s)
    Binding<LinearConstraint> CreateLinearConstraint(Formula f) {
        tuple<Matrix, Vector, VarList> info = magic_symbolic_extract<LinearConstraint>(f);
        auto constraint = CreateLinearConstraint(info.get<0>(), info.get<1>());
        return Binding<LinearConstraint>(constraint, info.get<2>());
    }

Then the usage would like:

    prog.Add(CreateLinearConstraint(A, b), vars); // numeric
    prog.Add(CreateLinearConstraint(A * x <= b)); // symbolic

NOTE: Edited Overview in the orignal post to use Option 3, `Create${CONSTRAINT}`.

hongkai-dai, 2017-04-19:
I think I still have the question here, is that the symbolic case of the proposal, we cannot do
```C++
prog.Add(A * x <= b);
```
to add a linear constraint, but the user has to tell the program that the constraint is linear, by using
```C++
prog.Add(CreateLinearConstraint(A * x <= b));
```
 On the other hand, in the current implementation, if we call
```C++
prog.AddConstraint(A*x<=b);
```
It knows the newly created constraint is linear. see https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/mathematical_program.cc#L584~L621. This is for `AddCost`, and @soonho-tri and I plan to add the similar idea to `AddConstraint`, such that 
`AddConstraint(const symbolic::Expression& e)` will add different types of constraint, depending on the structure of the expression `e` (linear, quadratic, polynomial, etc).

EricCousineau-TRI, 2017-04-19:
Ah, missed that, sorry!

Yes, this system could certainly handle that case using this bottom-up approach.
That could be covered by a generic factor, `CreateConstraint`, and then use a run-time check to place the constraint in the appropriate container. My thought was to have the layout as such:

* Specific invocations of `Add<C>(...)` place the constraint directly in the type-specified container, no run-time checks
* Specialize `Add<Constraint>(...)` to call `DoAddConstraintDynamic(Binding<Constraint>)`
    * This performs a run-time check to see if a constraint can be more efficiently placed (e.g., using `dynamic_cast<>` on the pointer and iterating through the types to see if they're addable, or explicitly querying some ID)

Then, in the generic symbolic form of `CreateConstraint` (or `CreateCost`), any constraint type can be returned, such as:

* `return CreatePolynomialCost(e) -> Binding<Constraint>(CreatePolynomialCost(coeffs), vars_vec)`
* `return CreateQuadraticCostWithMonomialToCoeffMap(...) -> Binding<Constraint>(CreateQuadraticCost(q, b), vars_vec)`
* `return CreateLinearCost(c, vars_vec)`

This does incur the additional run-time check, but I believe it will be negligible relative to the rest of the computations.

I was hoping that this could possibly simplify y'all's path, so please do let me know if it throws a wrench in your plans.

hongkai-dai, 2017-04-19:
I feel we should separate `AddCost` and `AddConstraint`, instead of combining them into `Add()` function. It is pretty clear what 
```C++
prog.Add(x+y<=1);
```
means. But I think the meaning of
```
prog.Add(2*x);
```
is not clear enough, to say it means "add 2*x to the cost". So I would propose we still keep two function signatures `AddCost` and `AddConstraint`.

EricCousineau-TRI, 2017-04-19:
Would you be OK with have the API as follows?

    Add(CreateConstraint(...)...)   -> AddConstraint(...)
    Add(CreateCost(...)...)  -> AddCost(...)
    AddCost(symbolic)
    AddConstraint(symbolic)

Alternatively, we would have some repetition (which would be fine, just want to make sure we give that the go ahead):

     AddConstraint(CreateLinearConstraint(...)...)
     AddCost(CreateQuadraticCost(...)...)

hongkai-dai, 2017-04-19:
I like your first API better
```
Add(CreateConstraint(...)...)   -> AddConstraint(...) // Non symbolic case
Add(CreateCost(...)...)  -> AddCost(...)  // Non symbolic case
AddCost(symbolic) // symbolic case
AddConstraint(symbolic) // symbolic case
```

EricCousineau-TRI, 2017-04-19:
Sorry for the verbosity, but just to recap, I've listed out our three alternatives. (I needed this to have my own internal debate :P ) I will update the main post once I flesh it out a tad more.

* **Alternative 1 (Current Winner)** - Yields some redundant and closely related options, but provides clarity when context may not be easily seen
* Alternative 2 - Minimal redundancy, grammatically and semantically
* Alternative 3 - Verbose but clear, grammatically redundant

TODO: Will need to consider options when using templates, such as being able to also specify `Add(CreateCost<QuadraticCost>(symbolic)...)` -> compile-time container placement.

## Alternative 1

* `Add(...)` only accepts a rigid set of non-symbolic overloads, such that it will error out if passed any symbolics directly.
* Will accept symbolics filtered through `Create${TYPE}(...)`
* `Add${TYPE}(...)` will forward directly to `Add(Create${TYPE}(...))`
* `Create${TYPE}` will NOT use parameter packs to minimize any chance of ambiguity

        Add(x.transpose() * Q * x);
            // Compilation error

        Add(CreateCost(x.transpose() * Q * x));
            // Compiles fine. Run-time container determination.

        AddCost(x.transpose() * Q * x);
            // Compiles fine. Run-time container determination.

        Add(CreateQuadraticCost(x.transpose() * Q * x));
            // Compiles fine. Compile-time container determination.

        AddCost(CreateQuadraticCost(x.transpose() * Q * x));
            // Compiles fine. Compile-time container determination.

This means that there is redundancy between two easily-reachable methods, `Add(Create${TYPE}(...))` and `Add${TYPE}(...)`, which may yield more options than needed. However, this is the lesser of many evils.

## Alternative 2

Instead of adding `AddCost(...)` and `AddConstraint(...)` with symbolic overloads, force the user to call `Create${TYPE}(...)` methods:

    Add(CreateCost(symbolic)...)
    Add(CreateConstraint(symbolic)...)

This makes things concise. The only drawback is, if you don't have a concise name, you may end up with something like:

    class Stuff {
    private:
        LinearConstraint x; // Poorly named constraint
    };
    ...
    prog.Add(x, vars); // Wait... Is this a cost or a constraint?

Granted, this is effectively the user's fault, but it also may be hard to clearly delineate overloads for `Add(...)`.

*NOTE*: Alternative 1 also does not prevent this from happening; however, users have the option of explicitly calling `AddConstraint(x, vars)` if they so choose.

## Alternative 3

Be verbose / grammatically redundant.

(We already said we did not want this, but still want to show it in the context for others if they wish to chime in.)

    AddConstraint(symbolic);
    AddConstraint(CreateCost(...));
    AddConstraint(CreateLinearConstraint(...));

EricCousineau-TRI, 2017-04-20:
Updated description. Going to actually do it now.

EricCousineau-TRI, 2017-04-24:
Will update PRs 1 and 2 by pulling some from this prototype branch:
https://github.com/RobotLocomotion/drake/compare/master...5ca783c#files_bucket

EricCousineau-TRI, 2017-04-26:
Added draft of Step 3 of PR Plan: https://github.com/EricCousineau-TRI/drake/pull/3/files

\cc @soonho-tri @hongkai-dai

EricCousineau-TRI, 2017-04-27:
Added draft of Step 4 of PR Plan: https://github.com/EricCousineau-TRI/drake/pull/4/commits

EricCousineau-TRI, 2017-04-27:
Added draft of Step 4 of PR Plan: https://github.com/EricCousineau-TRI/drake/pull/4/commits

\cc @soonho-tri @hongkai-dai

EricCousineau-TRI, 2017-04-28:
Instead of adding a separate issue for cleaning up stopgap crap, I've added it as Step 7 in the above PR Plan.

EricCousineau-TRI, 2017-04-28:
Two questions for y'all, @soonho-tri @hongkai-dai:

Do y'all have a preference on the `Create.*(Cost|Constraint)` names and signatures? Once concern I have is the potential for ambiguity / inconsistency:

#### Return Types
`Create.*(Cost|Constraint)` will return a `shared_ptr<C>` if there is no binding information, otherwise will return `Binding<C>`.

* **PRO**: This permits forms (a) `prog.Add(CreateLinearConstraint(A, b, lb, ub), vars)` and (b) `prog.Add(CreateLinearCost(lb <= A * x <= ub))`.
* **PRO**: `Add(...)` and `Add.*(Cost|Constraint)` will *always* return a `Binding<C>`.
* **CON**: See above.

Options:
1. Keep things as planned.
    * **PRO**: Simple and easy.
    * **CON**: May lead to confusion.
2. Enforce that `Create...` always return a `Binding<C>`, and extend the corresponding constructors.
    * **PRO**: Consistency, more work.
    * **CON**: Creation functions are explicitly coupled to bindings. You could not easily associate a `Constraint`/`Cost` with multiple distinct variable bindings.

My vote: (1) Keep things as planned.

#### Name
* **PRO**: The names are currently `Create.*`, to support grammar like `Add(CreateCost(...))`.
* **CON**:  Since `CreateCost(...)` may create a `shared_ptr`, it would be more in-line with the rest of C++ to use something like `MakeCost(...)`
    * Had been done before, but has been removed by PR 2.
* **CON**: As indicated above, the return-type may be different.

Options:
1. Keep things as planned.
    * **PRO**: Easy.
    * **PRO**: Distinguishes that the return type is sensitive to the inputs provided.
2. Change to `Make.*(Cost|Constraint)`
    * **PRO**: Analogous to STL.
    * **CON** (or **PRO**?): Would motivate that we keep as many creators as closed to `shared_ptr` as possible? Or could motivate that we make everything consistent.

My vote: (1) Keep things as planned.

soonho-tri, 2017-04-28:
For the return types, I don't like (1) but I'm afraid that (1) is the only option we can choose.

For the name, I think (1) is good.

hongkai-dai, 2017-04-28:
I think (1) is good. As long as `MathematicalProgram::Add(xxx)` always return a `Binding<C>` then I think we are good.

RussTedrake, 2017-04-29:
Sorry to be late to the party.  Just saw the proposed PR.

My initial reactions:
- the name MathematicalProgram::Add() , and the need for a "suggested best practices" feels clunky to me.  can you summarize for me why we don't just have `AddCost(Create.*Cost())` and `AddConstraint(Create.*Constraint())`.  It's a little more typing, but if it removes the redundancy in the interface and the need for a best practices doc, then I think it's warranted.  It would reduce your example above, with 5 potential call sites listed above to simply
```
  AddCost(x.transpose() * Q * x);
      // Compiles fine. Run-time container determination.

  AddCost(CreateQuadraticCost(x.transpose() * Q * x));
      // Compiles fine. Compile-time container determination.
```

presumably one could also do
```
  AddCost(CreateCost(x.transpose() * Q * x));
      // Compiles fine. Run-time container determination.
```
(but that is only allowed because we would like the ability to create costs/constraints outside of mathematical program... correct?)

- I like the idea of cleaning up `Create.*Constraint` to only return Bindings.  Agreed it's more work, and that your current proposal/PR is a perfect first step.  But could a second PR do further cleaning?  Is there any use case out there that needs to define a constraint independent of the binding to a variable?  Does that even make sense?

RussTedrake, 2017-04-29:
Another deficit in the current best practices proposal (at the top) is that AddCost can only return a Binding with the general constraint type.  So your suggested best practices could not use any derived class methods if the constraints were added with a symbolic type, e.g.
```
auto phi = prog.AddConstraint(A*x <= b)
...
phi->UpdateConstraint(newA,newb);
```

This might simply be an exception to your best practices, but it's worth considering.

Also, as implemented, I don't think `prog.AddCost(x.transpose()*Q*x)` will actually work, sinc e the type of the terms inside will be `Eigen::Matrix<symbolic::Expression>` not `symbolic::Expression`.  I would actually like to implement this method, too (and have it in a WIP branch) as @soonho-tri  and i did with https://github.com/RobotLocomotion/drake/blob/master/drake/systems/trajectory_optimization/direct_trajectory_optimization.h#L120

EricCousineau-TRI, 2017-04-30:
Glad to have your feedback!

BLOT, my takeaway action items:

1. Update the suggested interface, remove `Add()`
2. Simplify / remove best practices
3. Add examples of explicit type propagation
4. Add `cast<>` or `downcast<>` or `dynamic_cast<>` method to `Binding` directly.
5. Reduce syntactic-sugar overloads for `VectorXDecisionVariable`
6. Continue with PR plan things.

Verbose discussion stuff: (I kinda wish GitHub had a collapse-like functionality for extra stuff like this...)

#### `Add` vs `Add(Cost|Constraint)`
The main intent was to avoid redundancy in the words, but since there is more priority to get rid of redundant options, then I will reduce it to just `AddCost` and `AddConstraint`, and permit the redundant words to clean the interface.

#### Options to Downcast for `Add.*` methods with run-time container determination
In this case, if a user has foreknowledge on type to be added, then I believe the options are:

1. Use the explicit form of `Create.*(Cost|Constraint)`:

        auto phi = prog.AddConstraint(CreateLinearConstraint(A*x <= b));
        phi.constraint()->UpdateConstraint(newA,newb);

2. Perform an explicit downcast, as done for run-time determination [here](https://github.com/RobotLocomotion/drake/blob/d4aecde/drake/solvers/mathematical_program.cc#L445):

            auto phi = BindingDowncast<LinearConstraint>(prog.AddConstraint(A*x <= b));
            phi.constraint()->UpdateConstraint(newA,newb);

    * *Just realized that I mixed up downcast and upcast, thinking of the tree in reverse order - will incorporate that in the cleanup PR step...*

I personally prefer Option 1, as this would also avoid the need to do any run-time checks, as the user has already specified the constraint type.

However, some form of Option 2 could be used if the type may changed at run-time:

        Binding<Constraint> phi_;
        ...        
        phi_ = prog.AddConstraint(A*x <= b);
        ...
        // This RTTI check is dirty... Some other conditional should probably be used
        if (phi_.isa<LinearConstraint>()) {
            // This is less dirty, and we can ensure safety by asserting for non-nullness
            auto linear = phi_.cast<LinearConstraint>();
        } else if (phi_.isa<QuadraticConstraint>()) {
            ...
        }

#### Consistency of returning `Binding` objects
I can consolidate those interfaces. The main concern is that `Binding` objects also have the two and possibly more overloads if / when we wish to provide the ability to name the bindings, and this would be duplicated per `Create...` method.

I have not yet sifted through all use cases to create `Constraint` or `Cost` classes that are used in multiple `Binding`s, but given the immutability of these objects, I would like to keep that option open.

This could be done by adding another method, `Make.*(Cost|Constraint)Ptr`, which only returns a `shared_ptr`.
However, I will add this as optional for the time being. If this becomes too cumbersome, I can add it as a low-priority or backlog issue.

On that token, I will consider providing a (potentially copy-constructable) initializer to handle any `VectorXDecisionVariable` variants (such as `std::intializer_list`). I am thinking of doing something such as this (but more elegantly):

* [composition_ctor.cc](https://github.com/EricCousineau-TRI/repro/blob/d4bf765/cpp_quick/composition_ctor.cc)
    * [output](https://github.com/EricCousineau-TRI/repro/blob/d4bf765/cpp_quick/composition_ctor.output.txt)

This would reduce overloads to just one type. I mentioned this to @soonho-tri, and he somewhat likes the idea, but is concerned about permitting recursive initializer lists. I believe I have an idea for preventing this, and will pursue it.

EricCousineau-TRI, 2017-04-30:
On the Consistency on returning `Binding` objects, I believe I may change my opinion on this. I am still generally uncomfortable with the increased coupling between `Binding` and `Create.*(Cost|Constraint)`, and I feel that requiring these functions to return `Binding`s may artificially require more information than needed, especially if a `Binding` may mean more than just constraint and variables (e.g., context within a `MathematicalProgram`, such as its name).

This issue was initially meant to minimize that coupling, where before, if / when the Binding constructor is updated, we would have to update each individual `Add.*(Cost|Constraint)` method to reflect this, on top of all of the other changes that would need to be made.

If I enforce that each `Create.*(Cost|Constraint)` method enforce this, then that reintroduces the per-constraint/cost implementation coupling.
I'd like to possibly use template parameter packs, but (a) those can't transitively bind `initializer_list`s used for syntactic sugar, and (b) would have to be used everywhere, forcing all of those interface implementations to be available in the header file.

I would like to minimize as much input, even at the risk of inconsistent return types (which only occurs due to the fact that symbolic expressions have more information embedded in them). To this end, I would suggest that instead we have the following return types for `Create.*(Cost|Constraint)`:

* Numeric -> `std::shared_ptr<C>`
* Symbolic -> `std::pair< std::shared_ptr<C>, VectorXDecisionVariable >`.

This does have inconsistent return types, but they are not immediately usable as it would be with a `Binding`, so it would incentivize users to either (a) always pipe it to `MathematicalProgram::Add.*(Cost|Constraint)` to get the correct `Binding` (consolidating the `Binding` constructor coupling), or (b) know that they are getting additional information given how they formulate their constraints.

This also (most importantly) completely decouples constraint creation from `Binding` constructors, such that they can be changed.

For now, I will update my PR Plan to reflect this. However, if you still feel strongly that they should have a consistent return type, please let me know.

RussTedrake, 2017-04-30:
**RE:  Add vs Add[Cost|Constraint].**  I'm happy with using just `Add[Cost|Constraint]`, and removing `Add`.  I think all calls would be unambiguous without the extra characters, but feel that this just adds clarity.  Do @soonho-tri or @hongkai-dai agree?

**Re: 1x1 MatrixBase values.**  Thanks.  I think that's an obvious and helpful addition.  Again, ideally I would like it to work for the python bindings, too.  

**Re: Derived constraint classes and returning Bindings.**  Perhaps we need to flesh out the connection between Systems and Constraints (the other major use case) to really decide this?  My thoughts:  

I think we're agreed we want at least
```
Binding<Cost> MathematicalProgram::AddCost(symbolicExpression&)
Binding<Constraint> MathematicalProgram::AddConstraint(symbolicFormula&)
```
as the simplest interface to create the (correct type of) constraint, bind it to the variables, add it to the program, and return a Binding.  We need to support users adding constraints manually (no symbolic) for derived constraint classes, and support getting derived class pointers from the symbolic expression pathways, with a more streamlined interface that we have currently.  And we want to support binding constraints to variables outside of mathematical program (e.g. to elements of a context for system constraints).  

I'm not actually sure that we need to make new factories to accomplish this (e.g. your Create* methods).  We have a few class types floating around, and their constructors should do what we want?
1) `Constraint, LinearConstraint, BoundingBoxConstraint`, ...
2) `DecisionVariable, Context`, ...
3) `Binding<Constraint>, Binding<LinearConstraint>`, ...
4) `MathematicalProgram, System`, ...

**The thing that I am unclear on right now is whether we will instantiate a context templated on DecisionVariables, or somehow Bind a constraint to the context more directly?**

But for the mathematicalprogram use case, it seems like implementing, e.g.
```
MathematicalProgram::AddConstraint(Binding<LinearConstraint, DecisionVariable>)
```
and possibly also (as sugar)
```
Binding<Constraint> AddConstraint(shared_ptr<LinearConstraint>, DecisionVariable)
```
accomplishes the non-symbolic part of that.  Updating constraints could then be accomplished by e.g.
```
auto phi = make_shared<LinearConstraint>(A,b);
prog->AddConstraint(phi,x);
phi->UpdateConstraint(newA,newb);
```
I also wouldn't worry about making explicit downcast mechanisms yet, as long as this approach is not explicitly discouraged by some best-practices/style guide.  We can always add more sugar later.

Finally, for getting derived constraint classes from symbolic, it seems like a symbolic[Expression|Formula] exists fundamentally at the Binding level.  So we just need methods to get Binding<LinearConstraint,DecisionVariable> from the symbolicFormula.  A Binding is essentially a std::pair<Contraint, DecisionVariable>, so I wouldn't add that extra type flying around?

EricCousineau-TRI, 2017-04-30:
**Re^2: Derived constraint classes and returning Bindings**

> Perhaps we need to flesh out the connection between Systems and Constraints (the other major use case) to really decide this? 

I had only only spoken with @soonho-tri a bit about this, at the level that he had intended to separate out the symbolic parsing functionality (Soonho, please correct me if I am wrong). From this, I wanted to see if I can subsume at least a portion of that into this refactor. My assumption was the `System` framework may consume `Binding`s produced by the symbolic parsing, but I could definitely be wrong in this part.

I'll start reviewing a bit of #5297 (physical/non-physical state constraints) and #4998 (MTL / STL).
Are there any other discussions / design docs that would be pertinent?

> I'm not actually sure that we need to make new factories to accomplish this (e.g. your Create* methods). We have a few class types floating around, and their constructors should do what we want?

This is true when the factory method has a direct 1:1 with the type's constructor. However, it is most useful in the case of (a) symbolic variables (as you mentioned), and somewhat useful for the case of (b) different variants of the same type (e.g., (1) `AddLinearConstraint(Vector, double, double)`, (2) `AddL2NormCost`, etc.), which means, at present, the constructors do not do everything we want.

Example of extracted factory function: [CreateL2NormCost](https://github.com/EricCousineau-TRI/drake/pull/4/files#diff-5630193d3f8e5a2bf6e1f47b2ad2c716R88) (WIP)

Part (1) of (b) could be handled by constructor overloads, and we could move those creators there. For part (2) of (b), constructions that may need different names (`L2NormCost`, `QuadraticErrorCost`, ...), we could consider making those either (i) derived from the base type, possibly with more overloads (easiest, most succinct), or (ii) static factory methods of the respective constraint (`QuadraticCost::L2Norm(...)`).

For (b.2), I prefer a derived type, option (i).

> Finally, for getting derived constraint classes from symbolic, it seems like a symbolic[Expression|Formula] exists fundamentally at the Binding level.

If the `Create.*[Cost|Constraint]` methods could be eliminated for situation (b), then only (a) remains, where we  would then only be returning a constraint bound to its variables, which is what you mentioned above.

This would cause the C++ and Python APIs to deviate just a little. In C++, you would have:

    prog_.AddCost(make_shared<QuadraticErrorCost>(Q_, x_desired_), vars_);
    system_.AddConstraint(make_shared<LinearConstraint>(A_, b_), vars_);

while in Python:

    prog_.AddCost(QuadraticErrorCost(Q, x_desired), vars);
    system_.AddConstraint(LinearConstraint(A, b), vars);

which does seem acceptable. However, this would be the only way to add numeric constraints that are not already bound. I can't say I like this end-result interface more than the other, but it does clean up implementation and make the interface more consistent.

Would you like me to pursue this route, then?

> A Binding is essentially a std::pair<Contraint, DecisionVariable>, so I wouldn't add that extra type flying around?

True, but my fear is that we may have a desire to further overload whatever the symbolic factory functions are to return a "complete" Binding if we add an identifier such as a name (per #5827).
However, I do plan to accept "incomplete" `Binding`s (those without names), and assign the name when the binding is added, so the name could be assigned by a signature such as:

    AddCost(Binding<Cost>, string name = "")

In this case, I may consider throwing an error if the name is specified, but the Binding already has a non-default name, etc., which is doable.

Given this, I will avoid introducing the new type, `pair< shared_ptr<C>, Vars >` for now.

EricCousineau-TRI, 2017-05-01:
Per our discussion at the meeting, these are my takeaways:
(please let me know if there was anything that I did not capture accurately.)

* At present, there will be no other consumers of `Binding` aside from `MathematicalProgram`
    * It is fine to keep this type, and the factory methods, custom-tailored for or contained in `MathematicalProgram`.
* If a user knows what functional form of a symbolic expression they have, they may specify this with `Binding<${TYPE}>  AddCost<${TYPE}>(const ScalarOrMatrix<Expression>& e)` (or `AddConstraint`)
* There will be no methods of the form `shared_ptr<C> Create.*[Cost|Constraint](...)`. The user should call `make_shared<...>(...)` themselves.
    * There may be internal implementations for `Create.*[Cost|Constraint](const ScalarOrMatrix<symbolic::Expression>&)`, which would be called if the template-specialized `AddCost<C>(const ScalarOrMatrix<symbolic::Expression>&)` (or `AddConstraint`) is called.

I will go ahead and update the above description, example usages, and best practices.

\cc @RussTedrake @soonho-tri @hongkai-dai @jadecastro

EricCousineau-TRI, 2017-05-01:
Per a f2f with @soonho-tri, I plan to implement specialized types (e.g., `L2Norm`, `QuadraticError`) as types derived the base type, to enable a syntax such as:
```c++
prog_.AddCost(make_shared<L2NormCost>(A_, b_), vars_);
```

hongkai-dai, 2017-05-01:
Is `L2NormCost` a derived class of `QuadraticCost`?

EricCousineau-TRI, 2017-05-01:
Yup, such that it would be placed in the correct container.

hongkai-dai, 2017-05-01:
My concern is that we can have an explosion of derived class, if we go this way, as we can further have other derived class of QuadraticCost, for example, WeightedL2NormCost. L2NormCost is just a special form form of quadratic cost 0.5 * x' * Q * x + b' * x + c, where Q = Identity, b = 0, c = 0. Does this concern make sense?

EricCousineau-TRI, 2017-05-01:
Yes, that is most certainly a valid concern. In that case, may be a static method could be used, such as:
```c++
class QuadraticCost {
...
    static shared_ptr<QuadraticCost> MakeError(...);
    static shared_ptr<QuadraticCost> MakeL2Norm(...);
};
```

Alternatively, this could be made into external functions, such as `MakeQuadraticErrorCost(...)` and `MakeL2NormCost(...)`, but this would circle things back to the original design, and have a little bit if incongruence in that only special types have the `Make...Cost(...)` functional form.

hongkai-dai, 2017-05-01:
I think both these two designs (static method and external functions) are good, that they are just some functions to construct a `QuadraticCost` in a special form.

EricCousineau-TRI, 2017-05-01:
OK. To keep things somewhat decoupled (and keep the class definition less noisy), I will create separate `Make...Cost()` methods, possibly near the original classes. This would foster the ability to keep the syntax consistent if a user wanted to introduce an additional method or overload.

EricCousineau-TRI, 2017-05-01:
I've justed updated the description. Please let me know if you have any comments / concerns.

EricCousineau-TRI, 2017-05-01:
Did a quick test, template specialized overloads (e.g. `AddCost(symbolic...)`, `AddCost<C>(symbolic...)` work as expected:

* [tpl_method_overload.cc](https://github.com/EricCousineau-TRI/repro/blob/1356eca7f5dcd483a6aeed55d1dc56a0a80cf84f/cpp_quick/tpl_method_overload.cc#L138)
    * [example output](https://github.com/EricCousineau-TRI/repro/blob/1356eca7f5dcd483a6aeed55d1dc56a0a80cf84f/cpp_quick/tpl_method_overload.output.txt#L11)

Output snippet:
```
>>> c.AddConstraint(e_a).GetName()
compile-time { Binding<Constraint> }, run-time { Binding<AConstraint> }

>>> c.AddConstraint<AConstraint>(e_a).GetName()
compile-time { Binding<AConstraint> }, run-time { Binding<AConstraint> }
```

EricCousineau-TRI, 2017-05-12:
Updated description per f2f conversation yesterday.