RobotLocomotion/drake #5892: Bazel memcheck failing with shell scripts

m-chaturvedi, 2017-04-19:
Bazel memecheck fails when shell tests are run with it.
In particular when these tests are run with memcheck:
lcm_vector_gen_test
clang_format_includes_test

david-german-tri, 2017-04-20:
Can reproduce.  It looks like memcheck is instrumenting all the subprocesses and finding no errors, but is exiting 1 for some reason.  Definitely worth fixing, good catch.

david-german-tri, 2017-04-20:
From f2f: @m-chaturvedi says that `mv` and `cp` aren't working under valgrind, leading to the test failure.

david-german-tri, 2017-04-20:
From f2f: Are we really gaining anything by running a `sh_test` under memcheck anyway?  Probably not.

jwnimmer-tri, 2017-04-20:
Oops, yeah.  When `tools/valgrind.sh` is run on a `sh_test`, it should bail out and just `exec "$@"`.

Optionally in the future, it could also `export MAYBE_VALGRIND="$0"` first, so that `sh_test`s that cared could `$MAYBE_VALGRIND ./my_cxx_program` to re-enable valgrind on an wrapped program under test.  But probably we'll not care about that anytime soon.

stonier, 2017-05-04:
Closed with the PR submission linked above.

closed 2017-05-04 by stonier