RobotLocomotion/drake #5895: Should Modify DiagramBuilder::AddSystem() to take the name of the system as a parameter.

liangfok, 2017-04-20:
This is API sugar to avoid the common idiom of creating a system and then immediately naming it. For example, instead of:

```
auto pursuit = builder_->template AddSystem<PurePursuitController<T>>(*road_);
  pursuit->set_name("foo");
```
One would type:

```
auto pursuit = builder_->template AddSystem<PurePursuitController<T>>("foo", *road_);
```

Alternatively:

```
auto pursuit = builder_->template AddNamedSystem<PurePursuitController<T>>("foo", *road_);
```

liangfok, 2017-04-20:
@david-german-tri, I assigned this to you just to get your :+1: or :-1: . Please delegate to the appropriate person if this is something we should do.

david-german-tri, 2017-04-20:
:+1: for `AddNamedSystem`.  :-1: for changing the signature of `AddSystem`, since (a) some systems are self-naming and (b) it would touch a lot of code sites.  :-1: for fail-fast, since that would unnecessarily ban code that sets names after the fact.

Tagged as feature request, priority low, and reassigned to you.

liangfok, 2017-04-20:
Thanks @david-german-tri. I agree with your assessment.

jwnimmer-tri, 2018-09-13:
Over to @sherm1 for triage.

jwnimmer-tri, 2018-09-26:
I would like to +1 David's message above.  Since "AddSystem" already uses emplacement semantics, I do not think we should add a "name" argument to the front of it, even if we do it with overload.  The name argument could easily be ambiguous with the first constructor argument of the system being emplaced.  We should use a different method name such as "AddSystemWithName", if we want to offer an add-and-set_name sugar method.

sherm1, 2018-09-27:
I would prefer to enforce mandatory naming for systems as we're planning to do for input and output ports (with a rarely used, explicit "use default" option). That gives a much higher probability of good error messages and readable dot-diagrams, and reduces the number of annoyingly-similar method variants and API user decisions. My preference would be to do that using 
`AddSystem<MySystem>("name", MySystem_constructor_args...)`.

For backwards compatibility (during deprecation), we could still support the no-name variants except for the case where the first argument to the System constructor is a string -- those would have to be changed to be given an explicit name. Is that too violent?

jwnimmer-tri, 2018-09-27:
That a System within a Diagram must have a non-empty name is one thing.  We already have that, and we should keep it.  Forcing the user type in a string is another, and a strongly disagree.  We have already had a series of PRs to relax that constraint.

When we added that constraint a while ago, we went through and added literal string names for all the systems in every Diagram throughout Drake, it was in incredible bore and made lots of the code terrible to read and maintain.  I think, in many cases, naming the System using its typeid as a default name is plenty good enough.  I don't want to make people type `builder->AddSystem<Integrator>("integrator", ...)` as they assemble a Diagram.  Its redundant and noisy.

sherm1, 2018-09-27:
OK, I'm convinced -- thanks. I withdraw my support for mandatory user names for systems. In that case, on to #7128 so we can have nice defaults.

PR #14404 (2020-12-03): diagram builder: introduce AddNamedSystem sugar
Closes #5895.

This patch provides an potentially more fluent alternative to the common
AddSystem/set_name idiom.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14404)
commit aa4de299 (2020-12-03): diagram builder: introduce AddNamedSystem sugar (#14404)

* diagram builder: introduce AddNamedSystem sugar

Closes #5895.

This patch provides an potentially more fluent alternative to the common
AddSystem/set_name idiom.
closed 2020-12-03 by rpoyner-tri