RobotLocomotion/drake #5906: title capitalization in lcm_vector_gen is off for Rod2D

edrumwri, 2017-04-20:
lcm_vector_gen wants to capitalize "rod2d" to "Rod2d". Would it be too dumb to require the caller of lcm_vector_gen to specify the state vector name that they want?

jwnimmer-tri, 2017-04-20:
The name in which situation?  Classnames (type names) should be capitalized per google styleguide.  But maybe some other use of the name could be lowercased.

edrumwri, 2017-04-20:
The context is indeed a classname. The google style guide states, "Type names start with a capital letter and have a capital letter for each new word, with no underscores: MyExcitingClass, MyExcitingEnum." This doesn't indicate what should be done for "2D" or "3D", as examples.

jwnimmer-tri, 2017-04-20:
Oops, I thought the question was about the "R".  I see now its about the "D".

I think it'd be fine to allow the user to omit `--title` and instead specify `--camel` and `--snake` directly.  (I think `screaming_snake` is probably just `snake.upper()`, so should not be separately configurable.)

edrumwri, 2017-04-20:
Ok, I'll submit a PR on that.

jwnimmer-tri, 2017-05-05:
All issues must have owners; feel free to reassign elsewhere as needed (or close), etc.

sherm1, 2020-02-25:
Died for lack of interest.

closed 2020-02-25 by sherm1