RobotLocomotion/drake #5918: (Aesthetics) Consider moving Simulator execution semantics to explicitly facilitate stepper systems with no continuous state

EricCousineau-TRI, 2017-04-21:
This issue's high-level goal is to consider splitting off some of the execution semantics in `Simulator`, such that a user has the choice of placing a non-continuous system (`get_continuous_state()->size() == 0`) in either `Simulator` or a stepper compatible with soft real-time execution.

This is purely for aesthetic reasons (for now), since there are ways to use `Simulator` for (soft) real-time systems which contain only discrete states.

An example is @siyuanfeng-tri's [LcmDrivenLoop](https://github.com/RobotLocomotion/drake/blob/13b7cd9/drake/systems/lcm/lcm_driven_loop.h#L103) class, which exists primarily to (a) use timestamps provided from LCM (and should remain that way), but (b) leverages the execution / event-dispatch semantics for `Simulator` within a discrete state-only context, using a fixed-step integrator with trivial / unused continuous states to satisfy the requirements of `Simulator`.

Listed below are some alternatives to consider.

\cc @siyuanfeng-tri @sherm1 @edrumwri @rcory 

### Alternative 1

* `SystemStepperBase<T>`
    * Downstream classes must explicitly prohibit incompatible states at run-time (kind of ugly, but may work)
        * Example: Continuous states in a non-continuous stepper

    * Methods:
        * `protected: ctor(...)`
            * Make this abstract.
        * `Initialize()`
        * `ResetStatistics(...)`
        * `StepTo(...)`
            * Pseudcode:

                    bool sample_time_hit = false;
                    while (context_->get_time() < boundary_time || sample_time_hit) {
                        sample_time_hit = DoStepToNextEvent(boundary_time, actions);
                    }

        * `protected: virtual bool DoStepToNextEvent(double boundary_time, UpdateActions<T>&) = 0`
            * Contains the logic to access / mutate `UpdateActions<T>`
            * Will update time
            * Will return `sample_time_hit`
        * `protected: NextEvent DoHandleActions(...)` (same signature as `DoStepToNextEvent`)
            * Perform updates for non-continuous states, e.g.,
                * `HandleUnrestrictedUpdate(...)`
                * `HandleDiscreteUpdate(...)`
                * `HandlePublish(...)`

    * Data:

            struct NextEvent {
                double sample_time;
                double update_time; // not dt
                double publish_time;

                enum HitType { kSample, kUpdate, kPublish };

                std::pair<T, HitType> GetTimeAndType() {
                    // Pseudocode
                    return min((sample_time, kSample), (update_time, kUpdate), (publish_time, kPublish));
                }
            };

* `Simulator<T> : SystemStepperBase<T>`
    * Methods:
        * `DoIntegrationStuff(...)`
            * Will set context time appropriately (as it presently does)
        * `DoStepToNextEvent(...)`
            * Afterwards, integrate / query the next possible update time given the queued `UpdateActions<T>`

                    PauseIfTooFast();
                    NextEvent next_event = DoHandleActions(boundary_time, actions);
                    auto result = DoIntegrationStuff(boundary_time, actions, next_event);
                    ...
                    return sample_time_hit;

* `NoncontinuousSystemStepper<T> : SystemStepperBase<T>`, ...
    * Name is kinda ugly, but `DiscreteSystemStepper` seems wrong if we permit `Abstract` and `Publish` events
        * May be confused with a hybrid system too.
    * Methods:
        * `Initialize()`
            * Pukes if `get_continuous_state()->size() > 0`
        * `DoStepToNextEvent(...)`

                NextEvent next_event = DoHandleActions(boundary_time, ...);
                make_tuple(next_event_time, hit_type) = next_event.GetTimeAndType();
                bool sample_time_hit = (hit_type == kSample);
                // This could be passed some `SleeperBase` to permit different sleeping methods
                SleepUntil(next_event_time);
                return sample_time_hit;

*NOTE*: With this implementation, `LcmDrivenLoop<T>` could have `Simulation<T> stepper_` changed to `SystemStepperBase<T> stepper_`, such that it could be provided either

* `NoncontinuousSystemStepper<T>` - for nominal use, OR
* `Simulator<T>` - if, for whatever reason, you'd like to see a continuous-state simulation driven by Lcm time

### Alternative 2

* Avoid commonality and duplicate a subset of `Simulator`s functionality into `NoncontinuousSystemStepper<T>`.

### Alternative 3

* Keep on doin' what we're doin'

sherm1, 2017-04-24:
Before adding another layer of abstraction to this already-complicated class, I'd like to see whether we can use the existing Simulator effectively to solve the current problems. It is already very comfortable with discrete or continuous systems, and we expect that the same System class may use either formulation, depending for example on constructor arguments (e.g. RigidBodyPlant can use continuous or discrete contact models). We also expect that a system Diagram will be composed of an assortment of discrete, continuous, and hybrid subsystems. The Simulator as currently conceived supports the concept of a dynamic system advancing through time; whether that system is continuous, discrete, or hybrid does not matter and I am reluctant to elevate the significance of that relatively-minor modeling aspect to the level of a separate solver like `NoncontinuousSystemStepper`.

OTOH, as we discussed f2f we have been expecting to add additional types of solver as peers of Simulator, under the general category we've been calling "analysis". That could include for example a Monte Carlo solver, assembly analysis, equilibrium analysis, and so on. These are distinct from a Simulator in that they do not advance the System through time.

jwnimmer-tri, 2017-05-05:
All issues must have owners; feel free to reassign elsewhere as needed (or close), etc.

EricCousineau-TRI, 2017-10-04:
After working on #5629 and speaking with Siyuan,  I think it would be worth noting that `LcmDrivenLoop` does not reflect the same execution semantics as those that are present in `Simulator`.

Presently, `LcmDrivenLoop` disables `set_publish_every_time_step()`, and relies on `set_publish_on_every_received_message()` to then force a publish event every time the channel of interest receives a message, yielding an implicitly event-driven loop. Because of this, if any system declares a periodic event (publish or discrete update) that is not synchronized to the incoming message rate (which may not happen if the incoming time stamp from the sensor is not strictly progressing by a fixed `dt`), then it will "stutter" (if they are around the same `dt`, but with lead or lag) and "stack" the publishes (if the `dt` is significantly less than that of the driving LCM message).

This will happen silently, without the user's knowledge until they delve in and inspect this.

Ideally, it would be nice to (a) have a time-provider-driven stepper (where a time provider provides `now()` and `sleep(sec)`) and (b) still enable _explicitly_ event-driven loops (like an event trigger in Simulink).

Initially, (a) could be achieved with an "Alternative 4", which is to still compose a `Simulator` inside the stepper.
If merited, the execution semantics of `Simulator` could be moved to a common base class if `Simulator` permits functionality that would conflict with the time-provider-driven stepper.

\cc @naveenoid

EricCousineau-TRI, 2019-02-19:
I'm not sure if we've felt any significant pain points due to this specific issue. Other issues related to `LcmDrivenLoop` are better captured by #10714.

closed 2019-02-19 by EricCousineau-TRI