RobotLocomotion/drake #5923: team: robot locomotion group type:bug priority: high runMixInteger

sqhuang, 2017-04-22:
hello, i got a problem when i use the runMixIntergerSImpleForest example after read your paper.
I make my own example and it seem does't work not matter what all this parameters will be.
so i want you offer some help?
and what ub lb mean in the runMixedIntegerEnvironment?
Thank you.
here is my runMixedIntegerBeach.m

function runMixedIntegerBeach()
%RUNMIXEDINTEGERBEACH Summary of this function goes here
%   Detailed explanation goes here
 r = Quadrotor();

 

terrain = RigidBodyFlatTerrain();
terrain = terrain.setGeometryColor([0, 170, 255]'/256);
r = r.setTerrain(terrain);
r = addRobotFromURDF(r, 'beach.urdf');

start = [-5;-3;.1];
goal = [7; 2; 1.2];
seeds = [...
        [3,1,1.5]
        [3,-1,1.5]
         start';
         goal';
         ]';
        

lb = [-6;-4;-0.1];
ub = [10;3;3];

degree = 3;
n_segments = 6;
n_regions = 6;

runMixedIntegerEnvironment(r, start, goal, lb, ub, seeds, degree, n_segments, n_regions);

and it is my beach.urdf

<?xml version="1.0"?>
<robot name="office">

   <material name="Brown">
     <color rgba="83./255. 53./255. 10./255."/>
   </material>

   <material name="White">
    <color rgba="1 1 1"/>
   </material>

   <material name="Grey">
    <color rgba=".3 .3 .3"/>
   </material>

    <material name="Red">
     <color rgba="1 0 0"/>
    </material>

    <material name="Green">
     <color rgba="0 1 0"/>
    </material>

    <material name="Blue">
     <color rgba="0 0 1"/>
    </material>

   <link name="Hillton">
   	<inertial>
   		<mass value=".5" />
   		<origin xyz="0 0 0" rpy="0 0 0"/>
   	</inertial>

  	<visual>
      <origin xyz="0 0 0.5" rpy="0 0 pi/6" />
      <geometry>
        <box size="2 1 1" />
      </geometry>
      <material name="Brown"/>
    </visual>
    <collision>
      <origin xyz="0 0 0.5" rpy="0 0 pi/6" />
      <geometry>
        <box size="2 1 1" />
      </geometry>
      <material name="Brown"/>
    </collision>
   </link>

   <link name="Building">
   	<inertial>
   		<mass value=".5" />
   		<origin xyz="0 0 0" rpy="0 0 0"/>
   	</inertial>

  	<visual>
      <origin xyz="1.5 2 0.5" rpy="0 0 0" />
      <geometry>
        <box size="1 .5 1." />
      </geometry>
      <material name="Red"/>
    </visual>
    <collision>
      <origin xyz="1.5 2 0.5" rpy="0 0 0" />
      <geometry>
        <box size="1 .5 1." />
      </geometry>
      <material name="Red"/>
    </collision>
   </link>

   <link name="Vanke">
   	<inertial>
   		<mass value=".5" />
   		<origin xyz="0 0 0" rpy="0 0 0"/>
   	</inertial>

  	<visual>
      <origin xyz="7.5 -0.5 1.5" rpy="0 0 0" />
      <geometry>
        <box size="5.5 3 3" />
      </geometry>
      <material name="Green"/>
    </visual>
    <collision>
      <origin xyz="7.5 -0.5 1.5" rpy="0 0 0" />
      <geometry>
        <box size="5.5 3 3" />
      </geometry>
      <material name="Green"/>
    </collision>
   </link>
   
</robot>

rdeits, 2017-04-23:
`lb` and `ub` should be the lower bound and upper bound of the environment (in 3 dimensions). 

When you say it doesn't work, what exactly is happening?

sqhuang, 2017-04-24:
thank you, then i may not misunderstand the meaning of lb and ub.
it doesn't work like it just line two points(start and end) together.
[image]

sqhuang, 2017-05-01:
How can I do if console prints this:

> Integer solution solution summary
>  Problem status  : PRIMAL_INFEASIBLE
>  Solution status : UNKNOWN

> Interior-point solution summary
>   Problem status  : PRIMAL_AND_DUAL_FEASIBLE
>   Solution status : OPTIMAL

rdeits, 2017-05-01:
It looks like the initial mixed-integer program is infeasible, which is probably why the solution is incorrect. I would suggest adjusting your seed points and/or adding more of them.

sqhuang, 2017-05-02:
Thank you, I think I just find the solution. 
It was the size of the drone that cause the problem.
Nice work!

sqhuang, 2017-05-02:
...

closed 2017-05-02 by sqhuang