RobotLocomotion/drake #5924: Unable to Install Drake on OS X due to vtk5.rb.1 syntax error

liangfok, 2017-04-22:
# The Error

```
$ brew install autoconf automake bazel clang-format cmake doxygen gcc glib \
>   graphviz gtk+ jpeg libpng libtool libyaml mpfr ninja numpy python \
>   qt@4 qwt-qt4 scipy vtk5 wget

Error: vtk5: /usr/local/Homebrew/Library/Taps/robotlocomotion/homebrew-director/vtk5.rb:1: syntax error, unexpected <<
<<<<<<< HEAD
  ^
```

soonho-tri, 2017-04-22:
Please try:

```
brew untap homebrew-director/vtk5
brew tap homebrew-director/vtk5
```

It worked for me.

liangfok, 2017-04-22:
Odd. I get the following error:

```
$ brew untap homebrew-director/vtk5
Error: No available tap homebrew-director/vtk5.
```

soonho-tri, 2017-04-22:
Oouch, my bad. Try the following:

```bash
brew untap robotlocomotion/director
brew tap robotlocomotion/director
```

liangfok, 2017-04-22:
That worked. Thanks!!!

closed 2017-04-22 by liangfok

soonho-tri, 2017-04-22:
@liangfok and @patmarion : Just for your information, as far as I know, it is important to make non-destructive updates in `robotlocomotion/home-brew director` repository. If we do destructive updates on the repository, end users ' `homebrew` (sometimes) doesn't follow the updates in `robotlocomotion/director` tap.