RobotLocomotion/drake #5927: Unable to run drake-visualizer on OS X - WebKit Missing

liangfok, 2017-04-23:
# The Error

```
$ cd drake-distro
$ ./install/bin/drake-visualizer
dyld: Library not loaded: /usr/local/opt/qt/lib/QtWebKit.framework/Versions/4/QtWebKit
  Referenced from: /Users/liang/dev/drake-distro-1/build/./install/bin/drake-visualizer
  Reason: image not found
Abort trap: 6
```

Potentially related discussion: https://drakedevelopers.slack.com/archives/C0KDGF4V9/p1491533862131556

liangfok, 2017-04-23:
Maybe `drake-visualizer` should be updated to use QtWebEngine?

liangfok, 2017-04-23:
I just discovered a message about WebKit being removed for security reasons but can be manually added back:

```
$ brew info qt@4
cartr/qt4/qt@4: stable 4.8.7 (bottled), HEAD
Cross-platform application and UI framework
https://www.qt.io/
/usr/local/Cellar/qt@4/4.8.7_1 (3,058 files, 89.6MB) *
  Poured from bottle on 2017-04-21 at 11:09:25
From: https://github.com/cartr/homebrew-qt4/blob/master/qt@4.rb
==> Dependencies
Required: openssl ✔
Optional: dbus ✘, mysql ✘, postgresql ✘
==> Requirements
Build: git ✔
==> Options
--with-dbus
        Build with dbus support
--with-docs
        Build documentation
--with-mysql
        Build with mysql support
--with-postgresql
        Build with postgresql support
--HEAD
        Install HEAD version
==> Caveats
We agreed to the Qt opensource license for you.
If this is unacceptable you should uninstall.

Phonon is not supported on macOS Sierra or with Xcode 8.

WebKit is no longer included for security reasons. If you absolutely
need it, it can be installed with `brew install qt-webkit@2.3`.
```

Installing `qt-webkit@2.3` displays more disclaimers:

```
$ brew install qt-webkit@2.3
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 3 taps (caskroom/cask, homebrew/core, homebrew/science).
==> Updated Formulae
homebrew/science/libsequence               homebrew/science/plr                       lilv
homebrew/science/nixio                     homebrew/science/r-gui

==> Installing qt-webkit@2.3 from cartr/qt4
==> Downloading https://dl.bintray.com/cartr/autobottle-qt4/qt-webkit@2.3-2.3.4.sierra.bottle.1.tar.gz
######################################################################## 100.0%
==> Pouring qt-webkit@2.3-2.3.4.sierra.bottle.1.tar.gz
==> Caveats
This is years old and really insecure. You shouldn't
use it if you don't absolutely trust the HTML files
you're using it to browse. Definely avoid using it
in a general-purpose Web browser.

Also, video doesn't work.
==> Summary
🍺  /usr/local/Cellar/qt-webkit@2.3/2.3.4: 50 files, 25.5MB
```

Unfortunately, `drake-visualizer` still doesn't work:

```
$ cd drake-distro/build
$ ./install/bin/drake-visualizer
dyld: Library not loaded: /usr/local/opt/qt/lib/QtWebKit.framework/Versions/4/QtWebKit
  Referenced from: /usr/local/opt/vtk5/lib/vtk-5.10/libQVTK.5.10.dylib
  Reason: image not found
Abort trap: 6
```
It looks like vtk5 is searching for `QtWebKit` in `/usr/local/opt/qt/lib/QtWebKit.framework/Versions/4/QtWebKit` when it is actually located in `/usr/local/opt/qt-webkit@2.3`.

soonho-tri, 2017-05-03:
I'm not sure if it's related but I manage to make the OSX build working again thanks to @jamiesnape 's update on `vtk5` formula.

@liangfok , could you try it again? You might need to uninstall/untap qt/vtk formulas and then retap/install them back before testing it.

liangfok, 2017-05-03:
I uninstalled and reinstalled qt and vtk but ran into the following error:

```
$ ./install/bin/drake-visualizer
dyld: Library not loaded: /usr/local/Cellar/vtk5/5.10.1/lib/libvtkViews.5.10.dylib
  Referenced from: /usr/local/opt/vtk5/lib/vtk-5.10/libQVTK.5.10.dylib
  Reason: image not found
Abort trap: 6
```

The missing library is in `/usr/local/Cellar/vtk5/5.10.1/lib/vtk-5.10/libvtkViews.5.10.dylib`. Any idea what's wrong?

jamiesnape, 2017-05-03:
Did you `brew update`?

liangfok, 2017-05-03:
Yeah, here's what I see:

```
$ brew update
Updated 1 tap (caskroom/cask).
No changes to formulae.
```

jamiesnape, 2017-05-03:
Can you recompile `drake-visualizer`?

jamiesnape, 2017-05-03:
Ok, looks like a problem with the packages:
```
robotlocomotion/director/vtk5:
  * The installation was broken.
    Broken dylib links found:
      /usr/local/Cellar/vtk5/5.10.1/lib/libvtkCharts.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkGeovis.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkViews.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkInfovis.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkWidgets.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkVolumeRendering.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkHybrid.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkRendering.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkIO.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkGraphics.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkImaging.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkFiltering.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkCommon.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtksys.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libmpistubs.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkftgl.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkfreetype.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkViewsPythonD.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkInfovisPythonD.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkWidgetsPythonD.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkHybridPythonD.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkRenderingPythonD.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkGraphicsPythonD.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkImagingPythonD.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkIOPythonD.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkFilteringPythonD.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkCommonPythonD.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkPythonCore.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkGenericFiltering.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkproj4.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkverdict.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkexoIIc.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkNetCDF.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkDICOMParser.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkNetCDF_cxx.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libLSDyna.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkmetaio.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtksqlite.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libvtkalglib.5.10.dylib
        /usr/local/Cellar/vtk5/5.10.1/lib/libQVTK.5.10.dylib
```
I will have to fix that and rebuild.

soonho-tri, 2017-05-03:
Actually, I have the same problem when I run `./build/install/bin/drake-visualizer`. Sorry that I only tested building and running drake test binaries.

jamiesnape, 2017-05-04:
Try again now:
`brew remove vtk5 && brew untap robotlocomotion/director && brew tap robotlocomotion/director && brew install vtk5`

If you are unlucky, you may need:
`rm ~/Library/Caches/Homebrew/vtk5-5.10.1.{sierra|el_capitan|yosemite}.bottle.tar.gz `

liangfok, 2017-05-06:
Here's the error I'm seeing after executing the commands above:

```
$ ./install/bin/drake-visualizer
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
```

jamiesnape, 2017-05-08:
That would be a mismatch between Python library and Python interpreter. Another fix is needed.

jamiesnape, 2017-05-08:
New packages are available.

jamiesnape, 2017-05-09:
@liangfok @soonho-tri please test when you have a spare moment.

liangfok, 2017-05-09:
Alright, I should be able to test this afternoon. Should I just re-run the same commands as above? 

> brew remove vtk5 && brew untap robotlocomotion/director && brew tap robotlocomotion/director && brew install vtk5

jamiesnape, 2017-05-09:
Yes.

soonho-tri, 2017-05-09:
FYI, I'm testing it on my macs now.

soonho-tri, 2017-05-09:
It would be the best if we get a confirmation from @liangfok.

soonho-tri, 2017-05-09:
I have the following messages:
```bash
$ ./install/bin/drake-visualizer
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/soonhok/work/drake-distro/build/install/lib/python2.7/dist-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/Users/soonhok/work/drake-distro/build/install/lib/python2.7/dist-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/Users/soonhok/work/drake-distro/build/install/lib/python2.7/dist-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/Users/soonhok/work/drake-distro/build/install/lib/python2.7/dist-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
  File "/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtk/__init__.py", line 41, in <module>
  File "/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtk/vtkCommonCore.py", line 9, in <module>
ImportError: No module named vtkCommonCorePython
```

jamiesnape, 2017-05-09:
Ok. I will glad when we switch to VTK 7/8.

jamiesnape, 2017-05-09:
I don't even have a `/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtk` folder. Is it a symlink somewhere?

soonho-tri, 2017-05-09:
I think it's a symlink of `/usr/local/lib/python2.7/site-packages/vtk/`. Do you have this?

soonho-tri, 2017-05-09:
Is it possibly a residue of `vtk` (not `vtk5`)? I installed `vtk` accidentally but removed it later.

soonho-tri, 2017-05-09:
I removed the folder, `/usr/local/lib/python2.7/site-packages/vtk/`. But I have
```bash
ImportError: No module named vtk
```

Is it possible the `vtk5` bottle is created without Python support (i.e. using `--without-python`)? I'm build `vtk5` from source just to check if it makes things working.

soonho-tri, 2017-05-09:
I installed `vtk` using `--build-from-source`. It dropped files under `/usr/local/lib/python2.7/site-packages/vtk` directory. But the shared libraries in the dir have wrong pointers. For example:

```bash
$ otool -L /usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtk/vtkCommonPython.so
/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtk/vtkCommonPython.so:
	/tmp/vtk5-20170509-58587-1ijqstv/VTK5.10.1/build/bin/libvtkCommonPythonD.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/Python (compatibility version 2.7.0, current version 2.7.0)
	/tmp/vtk5-20170509-58587-1ijqstv/VTK5.10.1/build/bin/libvtkPythonCore.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/tmp/vtk5-20170509-58587-1ijqstv/VTK5.10.1/build/bin/libvtkCommon.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/tmp/vtk5-20170509-58587-1ijqstv/VTK5.10.1/build/bin/libvtksys.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.50.2)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.5.0)
```

As a result, I have the following error when I run `drake-visualizer`:

```bash
ImportError: dlopen(/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtk/vtkCommonPython.so, 2): Library not loaded: /tmp/vtk5-20170509-58587-1ijqstv/VTK5.10.1/build/bin/libvtkCommonPythonD.5.10.dylib
  Referenced from: /usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtk/vtkCommonPython.so
  Reason: image not found
```

jamiesnape, 2017-05-09:
Uninstall that `vtk5`, manually remove `/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtk` if it still exists, and then try the bottle again.

soonho-tri, 2017-05-09:
It works! Thank you!!
[image]

liangfok, 2017-05-09:
It works for me too! Thanks!!

closed 2017-05-09 by liangfok

justinej, 2018-06-02:
I ran into a similar problem with missing VTK files because I had installed VTK 8.1 using brew. I followed the instructions on this thread and did:
`brew uninstall vtk && brew tap robotlocomotion/director && brew install vtk@8.1`