RobotLocomotion/drake #5934: Systems within Diagrams should be assigned default names

jwnimmer-tri, 2017-04-24:
The restriction in #5866 is too easy for users of Diagram to violate, because the `System` constructor does not demand a name, so it's impossible to fail-fast, and you only find out at runtime that you forgot to bless you System with a name.

We should assign default names if the user left it blank, e.g., `NiceTypeName(*system) + "@" + pointer_to_string(system)`.

sherm1, 2017-04-25:
:+1: for a default name. I'd prefer a default that was more stable than a pointer, which can change from run to run (and is ugly!). Maybe just NiceTypeName + the subsystem number?

jwnimmer-tri, 2017-04-25:
My thought was to make it unstable _on purpose_ because it shouldn't be used for anything programmatic, or change any execution properties, etc.

sherm1, 2017-04-25:
I thought David's motivation was to have names for diagram drawings, for which some stability would be good. My experience with default node names in the past has been that they end up getting used a lot, either out of laziness or just because it is hard to think up better names than say, `gain_1` `gain_2` and `gain_3`. Is there a downside to making the defaults somewhat palatable?

jwnimmer-tri, 2017-04-25:
My starting position is that if you want a specific name on a drawing, debug text, etc., then you should set the name yourself.  My purpose for adding a default is for cases where nobody asks for the name ever, but the Diagram is still throwing at runtime right now from the empty names.

If we decided we wanted _stable_ default names, then I think they should just be `subsystem[0]`, `subsystem[1]`, etc., since `NiceTypeName` is only suitable for human output (its not stable).  Since the System names are likely to be abused for logic and not just diagnostics, I would prefer that they either be obviously unstable (so e.g. include the pointer as text) or obviously stable (`subsystem_0`).

jwnimmer-tri, 2017-04-26:
WIP at https://github.com/jwnimmer-tri/drake/tree/framework-default-system-names

PR #5953 (2017-04-27): Assign default names when adding Systems to a Diagram
Add `System::GetMemoryObjectName` to provide a diagnostic name, and use that within `DiagramBuilder` when a system is added without a name.

Closes #5934.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5953)
closed 2017-04-27 by jwnimmer-tri