RobotLocomotion/drake #5936: lcm_driven_loop_test Failure - Matrix size mismatch: (1 x 6 vs. 1 x 5)

liangfok, 2017-04-25:
# The Error

```
[ RUN      ] LcmDrivenLoopTest.TestLoop
/home/ubuntu/workspace/linux-xenial-clang-experimental/drake/systems/lcm/test/lcm_driven_loop_test.cc:131: Failure
Value of: drake::CompareMatrices(expected.transpose(), logger->data(), 1e-12, drake::MatrixCompareType::absolute)
  Actual: false (Matrix size mismatch: (1 x 6 vs. 1 x 5))
Expected: true
[  FAILED  ] LcmDrivenLoopTest.TestLoop (1104 ms)
```

Example log: https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-experimental/3902/

# Initial Prognosis

The unit test is using a _real_ LCM instance. Maybe it accidentally received a message that was not intended for it? Can the unit test be modified to use `DrakeMockLcm`?

siyuanfeng-tri, 2017-04-25:
hmm. probably a lcm flakyness. 
I can probably make it work with fake lcm, but i need to change stuff there.

liangfok, 2017-04-25:
BTW, it timed out in https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-experimental/3905/:

```
260/260 Test #172: lcm_driven_loop_test ...............................***Timeout  60.01 sec
```

jamiesnape, 2017-05-23:
This is blocking moving various Bazel memcheck builds to build cop.

jwnimmer-tri, 2017-05-23:
At worst, just `tags=['nomemcheck']` with a TODO though, right?

siyuanfeng-tri, 2017-05-23:
If it's blocking, feel free to take it off CI.

jamiesnape, 2017-05-23:
> At worst, just tags=['nomemcheck'] with a TODO though, right?

Yes, but for the various issues with combining tags.

jamiesnape, 2017-05-23:
Would anyone object to marking it `flaky` with a TODO?

siyuanfeng-tri, 2017-05-23:
Idk what flaky means, but sounds good to me...

jamiesnape, 2017-05-23:
https://bazel.build/versions/master/docs/be/common-definitions.html#common-attributes-tests:
> Marks test as flaky.
>
> If set, executes the test up to 3 times before being declared as failed. By default this attribute is set to 0 and test is considered to be stable. Note, that use of this attribute is generally discouraged - we do prefer all tests to be stable.

m-chaturvedi, 2017-06-13:
We have disabled the test for a few sanitizers, we would want to re-enable them after we've sorted this issue. https://github.com/RobotLocomotion/drake/pull/6327

hongkai-dai, 2019-03-25:
@siyuanfeng-tri could you update the status of this issue?

jwnimmer-tri, 2019-03-26:
I will dispose of this issue forthwith.

jwnimmer-tri, 2019-04-19:
Closed as of #11281.

PR #11281 (2019-04-22): systems/lcm: Deprecate lcm_driven_loop
Closes #5936.  Closes #10714.

---

**If you are a Drake user and would like to adjust your code to the new APIs, please read on for details.**

Drake has deprecated (and will eventually remove) the LcmDrivenLoop class, which organized the execution and message flow of an LCM-triggered control loop.

LcmDrivenLoop presents challenging problems with threading, mutexes, and condition variables because it assumes the LCM service in handled by a separate thread (no longer true as of #11166), and it is insufficiently general in the first place (#10714).

For a [... cut]
closed 2019-04-19 by jwnimmer-tri