RobotLocomotion/drake #5940: Sparse documentation in SignalLogger

samzapo, 2017-04-25:
It's not clear what the signal logger does or how to use it:
https://github.com/RobotLocomotion/drake/blob/master/drake/systems/primitives/signal_logger.h

Additionally, the signal logger test code documentation does not demonstrate how to use the logger.  This would be a great place to add a brief tutorial of how to use the signal logger---even if it doesn't dive very deep.
https://github.com/RobotLocomotion/drake/blob/master/drake/systems/primitives/test/signal_logger_test.cc

I suggest adding 3 types of documentation:
1) High level description of how the code relates to drake (e.g., "this logs the output of that")
2) C++ description of what's happening (e.g., "a pointer to a LeafSystem output port is used to construct the class")
3) (optional) Scientific description of what the code does (i.e., math or algorithmic explanation)

jwnimmer-tri, 2017-05-05:
All issues must have owners; feel free to reassign elsewhere as needed (or close), etc.

siyuanfeng-tri, 2017-05-05:
https://github.com/RobotLocomotion/drake/pull/6044
@samzapo see if you are happy with this. there is really nothing to it, so we had the doc pretty minimal, sorry for that.

closed 2017-09-18 by siyuanfeng-tri