RobotLocomotion/drake #5948: Future of Matlab bindings post-Cmake

rdeits, 2017-04-26:
My understanding is that the Bazel build won't have Matlab mex file support, and that the Cmake build will eventually be switched off. Does that imply that the existing Matlab interface will eventually stop working and/or be removed?

I'm not using any of those files, but this question has come up several times in the lab, and it would be nice to have a clear answer that I can refer people to.

RussTedrake, 2017-04-27:
The goal is to zap any places in the drake/ tree where we are building things with BOTH bazel and cmake.  bazel builds will export their targets in reasonable ways, including a FindDrake.cmake yada.  That way external projects can link against drake w/ cmake.  

The current plan for mex files is to leave cmake in the drake tree for add_mex functionality, but have it link against the bazel targets in the same way that an external project would.

rdeits, 2017-04-27:
Ok, good to know, thanks.

closed 2017-04-27 by RussTedrake