RobotLocomotion/drake #5956: Valgrind version for a ignition_rndf_test

m-chaturvedi, 2017-04-27:
ignition_rndf_test ( formerly ignition_math_test )  seems to fail with the current version ( 3.11.0 ) of valgrind,  but if we use 3.12.0 it passes.  

To reproduce :
`bazel test  --config=memcheck --verbose_failures //drake/automotive/maliput/rndf:ignition_rndf_test `

with the appropriate version of valgrind in valgrind.sh.  
This is the reported issue:
https://github.com/google/protobuf/issues/3006

stonier, 2017-04-27:
What is the root cause of it failing in 3.11.0? I can't see anything in the report issue that makes it obvious that it is valgrind's fault in the reported issue - just that the problem magically disappeared in 3.12.

jamiesnape, 2017-04-27:
@m-chaturvedi Can you post the output of the test here?

jwnimmer-tri, 2017-04-27:
Machine code emulation was broken in 3.11.0.  It advertised as a CPU with RDRAND but didn't actually implement RDRAND.  https://bugs.launchpad.net/ubuntu/+source/valgrind/+bug/1501545

m-chaturvedi, 2017-04-27:
The output is practically the same as in the link I'd posted, here's it:
https://gist.github.com/m-chaturvedi/75c14e294e0e0e5116683bd4dd0f956a

jwnimmer-tri, 2017-04-28:
Besides the valgrind error, this points out that `ignition-math` has C++ global constructors (non-POD globals) e.g. https://bitbucket.org/ignitionrobotics/ign-math/src/5d674fbadeaf04fca7f588b1b2d9cd46a9785324/src/Rand.cc?at=default&fileviewer=file-view-default#Rand.cc-33.

Global initializers are a no-no in C++ due to the static initialization fiasco -- see https://google.github.io/styleguide/cppguide.html#Static_and_Global_Variables for the rule and https://isocpp.org/wiki/faq/ctors#static-init-order for the why.

The correct way to do it is something like https://github.com/RobotLocomotion/drake/blob/3f349759e36dd6917ddeb937d97e57a31c307ce1/drake/common/symbolic_variable.cc#L23, where the global variable is a function-local static in C++, and has no destructor.

stonier, 2017-05-01:
As far as I can see, to move forward we need to either:

1. Disable this test OR
2.  Require everyone to have an updated valgrind

The second I'd hazard is impractical. @jwnimmer-tri with your software core hat on - confirmed?

Sent an issue upstream wrt to the static initialisation topic.

jwnimmer-tri, 2017-05-01:
Since most _users_ will not run valgrind, but rather only Drake Developers will run it, it's at least somewhat plausible that we could require a newest valgrind.  Certainly within TRI this would be easy.

However, there is another solution you didn't mention: (3) fix `ignition-math` so that it doesn't seed randomness unless someone asks it to, and so that it doesn't use non-POD global variables.

stonier, 2017-05-01:
> However, there is another solution you didn't mention: (3) fix ignition-math so that it doesn't seed randomness unless someone asks it to, and so that it doesn't use non-POD global variables.

I already issued a ticket on ignition-math [#69](https://bitbucket.org/ignitionrobotics/ign-math/issues/69/redirect-static-initialisation-of) but wasn't sure if that would resolve valgrind's issues with it or not. If this does work, then should be simple enough to fix there without breaking api as they already have a `Seed()` function within which they can nest the initialisation.

jwnimmer-tri, 2017-05-01:
No matter the ign-math APIs, it would always fixable without API changes.  I cannot see any reason why valgrind would still complain after that fix is made.

stonier, 2017-05-01:
If they'd made that variable public and used it elsewhere then it would have disrupted other people's code.

I'll bump them this week to address the issue. @m-chaturvedi would you like to disable it for now so you can move on and I'll update here when it's fixed upstream?

jwnimmer-tri, 2017-05-01:
Oh, you're right.  Mutable public global variables didn't even cross my mind, because _who does that!_.

m-chaturvedi, 2017-05-01:
@stonier Sure.

stonier, 2017-05-03:
@m-chaturvedi Can you test with the branch they have in [ign-math/pull-request/165](https://bitbucket.org/ignitionrobotics/ign-math/pull-requests/165/construct-on-first-use-in-rand-class/diff)?

m-chaturvedi, 2017-05-03:
I checked using the branch issue_69_math3 locally on my machine.  The test //drake/automotive/maliput/rndf:ignition_rndf_test seems to pass.  When they merge it to `default` we shouldn't have this problem.

stonier, 2017-05-14:
@m-chaturvedi that ign-math pull request is now in. You'll have to update the ign-math version/hash drake is pulling in (Bazel and CMake) to get it and then re-enable it in the memcheck test.

I asked them over in [ign-math/pull-requests/165](https://bitbucket.org/ignitionrobotics/ign-math/pull-requests/165/construct-on-first-use-in-rand-class/diff) whether they'll make a new tagged release. You might want to follow up with that - find something that works to make sure you get the new commits.

m-chaturvedi, 2017-05-17:
The issue was addressed in : 
https://github.com/RobotLocomotion/drake/pull/6112

closed 2017-05-17 by jamiesnape