RobotLocomotion/drake #5974: Ensure that Scalar's used within Eigen::Matrix<> containers use the correct code paths upon resizing

EricCousineau-TRI, 2017-04-29:
@jwnimmer-tri, @soonho-tri, and I did some digging, and realized that `Eigen::Matrix<>` will not play well with `std::string` because of it's [Short String Optimization](http://stackoverflow.com/questions/10315041/meaning-of-acronym-sso-in-the-context-of-stdstring/10319672#10319672). Per Jeremy's words:

> For short strings, the data pointer points back into an offset of the `this` pointer, and there is no heap allocation.
> But when you _reallocate_ where `this` is, the self-pointer is no longer valid.

## Takeaway
We need to ensure that we use the correct code paths when an `Eigen::Matrix<>` needs to be resized when the `Scalar` type may not be [Trivially Copyable](http://en.cppreference.com/w/cpp/concept/TriviallyCopyable). The options are:

1. Ensure that the `Scalar` type is indeed trivially copyable. In this case, could mean:
    * Making any `std::string` (or other things that do weird pointer arithmetic) members either (a) pre-allocate via `resize` to avoid SSO, (b) change it to a `std::string*` and allocate it on the heap s.t. the pointer is trivially copyable, or (c) use a primitive type, such as `char[N]` or `std::array<char, N>` with a maximum size.
2. Update Eigen to have an optional flag to use copy/move constructors with placement `new` / `delete`, to be robust to stuff like this. This would hurt efforts such as #5785, but would definitely be a good thing to contribute towards Eigen.
3. Try to ensure that `conservativeResize` calls the copy constructor when reallocating. In most cases, it may use placement `new` and `delete`. However, (I think) in others, it may allocate a temporary and do a block assignment (with implicit construction).
    * This enables non-trivial types. This seems to be triggered (at least in a test case) with `conservativeResize(int, int)`. On the other hand, `conservativeResize(int)` seems to fail.

## Notes
This does not seem to affect `AutoDiff...` stuff, since it uses Eigen containers which do not seem to do any SSO-like pointer ~hackery~ magic.

## Repro Steps
Directory: [eigen_resize_dtor](https://github.com/EricCousineau-TRI/repro/tree/issue/eigen_override/bug/eigen_resize_dtor)
Branch: `issue/eigen_override`
Files:
* `conservative_resize_issue.cc` - Shows `AppendToVector` being implemented with `conservativeResize(int)` and `conservativeResize(int, int)`.
* `output.txt` - The output from executing `./repro.sh`.

Cases: (and SHA)
* Memory leak with short `std::string` scalars: [b1085a0](https://github.com/EricCousineau-TRI/repro/tree/b1085a0/bug/eigen_resize_dtor)
* No leak with long `std::string` scalars: [2ffbfe8](https://github.com/EricCousineau-TRI/repro/tree/2ffbfe8/bug/eigen_resize_dtor)
* No leak with `VectorXd` scalars: [fa68f30](https://github.com/EricCousineau-TRI/repro/tree/b1085a0/bug/eigen_resize_dtor)

Thus far, these steps have only been confirmed to fail on `clang version 3.9.1-svn288847-1~exp1 (branches/release_39)`, on Ubuntu 16.04 LTS.

EricCousineau-TRI, 2017-04-30:
One additional note: I started investigating this because I was encountering hard crashes when using `conservativeResize(int, int)` on my machine.

EricCousineau-TRI, 2017-05-01:
Per f2f with Soonho and Jeremy, the main thing to distill is that an object's representation must be invariant of its location in memory, hence the above edit from "Trivially Copyable" to "`memcpy` Movable".

jwnimmer-tri, 2017-05-02:
FYI the issue title "Leak memory" is a bit misleading.  "Crash and burn with extreme UB prejudice" is more like it.

EricCousineau-TRI, 2017-05-03:
Tentatively closing this issue per @soonho-tri's PR #5983.

We can reopen or reference this issue if we encounter any other scalar types in the future.

closed 2017-05-03 by EricCousineau-TRI

EricCousineau-TRI, 2017-05-05:
Posted on the Eigen forum to see what other solutions may be available (since there seems to be something like `smart_copy` that either does `memcpy` or `std::copy`):
[Correct Method to use Non-Memcpy-Movable Types as Scalars?](https://forum.kde.org/viewtopic.php?f=74&t=140169)

jwnimmer-tri, 2023-03-06:
The underlying [Eigen problem](https://gitlab.com/libeigen/eigen/-/issues/1432) of reallocation being ignorant of trivially copyable types was fixed in [this commit](https://gitlab.com/libeigen/eigen/-/commit/863580fe881c32af82eb106817e71dc560d9e775).

The fix was released as of Eigen 3.4.0.

(It was targeted for the Eigen 3.3.x branch but never made it, at least not as of Eigen 3.3.9.)