RobotLocomotion/drake #5975: Need to Remove LCM Dependency from AutomotiveSimulator

liangfok, 2017-04-29:
To allow `AutomotiveSimulator` to be used in environments that do have LCM, we should remove `AutomotiveSimulator`'s dependency on LCM.

One idea is to split the current `AutomotiveSimulator` class into a base class (`AutomotiveSimulatorBase`) that does not depend on LCM and a child class (e.g., `AutomotiveSimulatorLcm`) that does.

Another idea is to extract the objects within `AutomotiveSimulator` that depend on LCM into helper methods that add them to `AutomotiveSimulator`'s `Diagram`.

The following `AutomotiveSimulator` member variables depend on LCM:
* `std::unique_ptr<lcm::DrakeLcmInterface> lcm_{};`
* `systems::rendering::PoseBundleToDrawMessage* car_vis_applicator_to_draw_{};`
* `systems::lcm::LcmPublisherSystem* car_vis_applicator_lcm_publisher_{};`

jwnimmer-tri, 2017-04-29:
Again, once you remove LCM from the picture, what remaining value is there in the class?  It's just making a Diagram and adding a few systems, which when you don't have the visualization baggage is just a simple few lines of code that can be rewritten from scratch for the new use case.  Not everyone who wants to experiment with automotive systems should have to do it via `AutomotiveSimulator` -- it's a purpose-built demo program, not a One True Middleware.

liangfok, 2017-04-29:
I'm not sure if all that's left after removing LCM is indeed just a few lines of code, but will investigate.

RussTedrake, 2017-04-30:
i actually think it is a very useful middleware, even without LCM.  it's a common place for instantiating the different type of vehicle models, binding them to road geometries, etc. 
i had to do exactly this in my WIP branch to get trajectory optimization working.

jwnimmer-tri, 2017-04-30:
That's fine, but none of those things help Liang on the TRI-internal project he's proposed this for.

From your prior comments, what you did was turn off LCM instantiation, but left it as a build-time dependency.  This PR proposes removing it as a build-time dependency, at which point only about 10 lines of code are left in the class.

liangfok, 2017-05-02:
Upon further investigation, simply removing the LCM dependency is not sufficient for my purposes. I shall thus proceed as @jwnimmer-tri recommended and not use `AutomotiveSimulator`. Since there is otherwise no immediate need for this feature, I will close this issue.

closed 2017-05-02 by liangfok