RobotLocomotion/drake #5979: What is 'MIMOCascade_2856590/system1/DrakeSys'

GuiyangXin, 2017-04-29:
Hello guys,
I cannot run 'runAtlasWalking.m' or 'runAtlasBalancingPerturb.m'. The same error prompted as follows
````
Error using getInitialState (line 43)
Error reported by S-function 'DCSFunction' in
'MIMOCascade_2856590/system1/DrakeSys':

DrakeSystem S-Function: error MATLAB:Java:GenericException
in MATLAB callback.
See additional debugging information above

Error in sl (line 15)
[varargout{1:nargout}]=feval(varargin{:});

Error in Simulink.BlockDiagram.getInitialState (line 26)

Error in getSimulinkStateStructure (line 13)
x_struct = Simulink.BlockDiagram.getInitialState(mdl);

Error in DynamicalSystem/stateVectorToStructure (line 657)
        obj.structured_x =
        getSimulinkStateStructure(mdl.name);

Error in DynamicalSystem/simulate (line 39)
  x0 = stateVectorToStructure(obj,x0,mdl);

Error in Biped/simulateWalking (line 30)
ytraj = simulate(sys, [0, T], x0, options);

Error in Biped/runWalkingDemo (line 52)
[ytraj, com, rms_com] =
obj.simulateWalking(walking_plan_data, walking_options);

Error in runAtlasWalking (line 31)
r.runWalkingDemo(walking_options);
````
But I can run 'runAtlasJumping.m' and 'runDRCPracticeTerrain.m'. And there are several files like 'MIMOCascade_1723838.slx.autosave' in the example folders after I run these programs for several times.

hongkai-dai, 2017-04-29:
`MIMOCascade_1723838.slx.autosave` is the simulink block creeated automatically by drake simulation. Your problem is that the initial state cannot be computed successfully. Could you add a break point in `getInitialState` function inside `RigidBodyManipulator.m`, step into this function, and check which line throws the error? Also, do you have snopt on your machine?

GuiyangXin, 2017-04-30:
Thank you for your reply. First, I think I have installed SNOPT. I followed this guidance [https://github.com/openhumanoids/oh-distro-private](url) to install the software. Sometimes I can see ' Calling addpath_snopt' that is prompted in Matlab's command window. 
Secondly, I put some break points in `getInitialState` inside `RigidBodyManipulator.m`, but the error still generated without stopping at these break points. I post the total error report here
````
Error using bipedControllers.BipedPlanEvalAndControlSystem/output (line 75)
Java exception occurred: 
java.io.IOException: LCM Decode error: bad fingerprint
	at drake.lcmt_qp_controller_input.<init>(lcmt_qp_controller_input.java:110)
	at drake.lcmt_qp_controller_input.<init>(lcmt_qp_controller_input.java:104)

Error in FeedbackSystem/getOutputs (line 162)
        y2=output(obj.sys2,t,x2,sat2(obj,y1));

Error in FeedbackSystem/getInitialStateWInput (line 101)
      [y1,y2]=getOutputs(obj,t,x,u);

Error in getInitialState (line 43)

Error in sl (line 15)
[varargout{1:nargout}]=feval(varargin{:});

Error in Simulink.BlockDiagram.getInitialState (line 26)

Error in getSimulinkStateStructure (line 13)
x_struct = Simulink.BlockDiagram.getInitialState(mdl);

Error in DynamicalSystem/stateVectorToStructure (line 657)
        obj.structured_x = getSimulinkStateStructure(mdl.name);

Error in DynamicalSystem/simulate (line 39)
  x0 = stateVectorToStructure(obj,x0,mdl);

Error in Biped/simulateWalking (line 30)
ytraj = simulate(sys, [0, T], x0, options);

Error in Biped/runWalkingDemo (line 52)
[ytraj, com, rms_com] = obj.simulateWalking(walking_plan_data, walking_options);

Error in runAtlasWalking (line 31)
r.runWalkingDemo(walking_options);
Error using getInitialState (line 43)
Error reported by S-function 'DCSFunction' in
'MIMOCascade_3924621/system1/DrakeSys':

DrakeSystem S-Function: error MATLAB:Java:GenericException
in MATLAB callback.
See additional debugging information above

Error in sl (line 15)
[varargout{1:nargout}]=feval(varargin{:});

Error in Simulink.BlockDiagram.getInitialState (line 26)

Error in getSimulinkStateStructure (line 13)
x_struct = Simulink.BlockDiagram.getInitialState(mdl);

Error in DynamicalSystem/stateVectorToStructure (line 657)
        obj.structured_x =
        getSimulinkStateStructure(mdl.name);

Error in DynamicalSystem/simulate (line 39)
  x0 = stateVectorToStructure(obj,x0,mdl);

Error in Biped/simulateWalking (line 30)
ytraj = simulate(sys, [0, T], x0, options);

Error in Biped/runWalkingDemo (line 52)
[ytraj, com, rms_com] =
obj.simulateWalking(walking_plan_data, walking_options);

Error in runAtlasWalking (line 31)
r.runWalkingDemo(walking_options);

````

hongkai-dai, 2017-05-02:
@rdeits , I am not very familiar with these LCM erros, have you seen them before?

rdeits, 2017-05-02:
It could be https://github.com/lcm-proj/lcm/issues/46 which would indicate an out-of-date version of LCM.

GuiyangXin, 2017-05-03:
Thank you all. But how can I fix it? How can I update the version of LCM? Is the following code enough to fix it?
````
wget https://github.com/lcm-proj/lcm/releases/download/v1.3.1/lcm-1.3.1.zip
unzip lcm-1.3.1.zip
cd lcm-1.3.1
./configure
make
sudo make install
````
Do I need to build Drake again? Forgive me for some naive question. I am a beginner for computer science. Thanks.

rdeits, 2017-05-03:
Actually, I'm not sure what it could be. From your previous comment, I thought you were trying to build the `oh-distro` project, but now I see that you're just using Drake. Drake should already be building the correct version of LCM, so downloading it should not have been necessary (sorry for the confusion). 

Are you running Drake master?

GuiyangXin, 2017-05-03:
Yes, I followed the guidance in `oh-distro-private`. I guess I am running the Drake master branch. But how can I recognise it? I think it should be based on which branch I cloned from github.

rdeits, 2017-05-03:
You can check with `git status` which should show `on branch master`. How recently did you clone drake?

GuiyangXin, 2017-05-04:
I am working on LittleDog. So I have just made commit in the directory of LittleDog. When I check using` git status` under the directory of /oh-distro-private/software/drake/drake, it responses `HEAD detached at ce26d71`. I cloned Drake in November 2016. And I have never update it because I don't know how to make it up-to-date. Do I need to clone the entire package again? And compile it? Or is there an alternative way that can keep my repository being the same with the version on this site RobotLocomotion with the help of git? Thank you very much.

rdeits, 2017-05-04:
Ah, yes, that's a pretty old version. I didn't realize you were using the version of Drake inside of `oh-distro-private`. Unfortunately, I don't think that a more recent version of Drake will work with `oh-distro`. Perhaps you could just build Drake by itself http://drake.mit.edu/from_source.html instead? 

As for keeping up to date with git, the general answer is that you need to use `git pull` to bring in the changes from this repository, and then you need to run `make` again to compile it. But the better answer is that learning `git` by posting issues here isn't going to be very productive, and I'd strongly suggest reading a real git tutorial like: https://git-scm.com/docs/gittutorial or https://try.github.io/levels/1/challenges/1

GuiyangXin, 2017-05-04:
OK, thank you so much Robin. I am going to have a try. I think this issue can be closed. Thank you all.

closed 2017-05-05 by RussTedrake