RobotLocomotion/drake #6013: MATLAB testMathematicalProgram fails during coverage builds

jamiesnape, 2017-05-03:
https://drake-cdash.csail.mit.edu/testDetails.php?test=12242407&build=573070
https://drake-cdash.csail.mit.edu/testDetails.php?test=12176080&build=572899
```
Matlab test  output:

                            < M A T L A B (R) >
                  Copyright 1984-2015 The MathWorks, Inc.
                   R2015b (8.6.0.267246) 64-bit (glnxa64)
                              August 20, 2015

 
To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.
 

	Academic License

Undefined variable "py" or class "py.pydrake.solvers.mathematicalprogram.MathematicalProgram".

Error in testMathematicalProgram (line 17)
prog = py.pydrake.solvers.mathematicalprogram.MathematicalProgram();

 

CMake Error at /usr/local/share/cmake-3.7/Modules/MatlabTestsRedirect.cmake:104 (message):
  [MATLAB] TEST FAILED Matlab returned 1
```
FYI @RussTedrake.

RussTedrake, 2017-05-03:
hrmm.  don't know why that build would be different.  i've manually added the PYTHONPATH to the cmake output in build/install/lib/python2.7/...  
@EricCousineau-TRI -- any ideas?

jamiesnape, 2017-05-03:
I see why:
```
# Disable Python because SWIG generation fails
cache_append(DISABLE_PYTHON BOOL ON)
```

https://github.com/RobotLocomotion/drake-ci/blob/master/driver/configurations/coverage.cmake#L70-L71

We do not use SWIG now so I will try removing that line. That test also needs guarding with something. I will PR a fix.

EricCousineau-TRI, 2017-05-03:
*NOTE*: Was writing this out before Jamie solved it (whoop!), but will still post as relevant item for Slack convo.

Writing it out (namely for myself) to see why it is build-dependent:

Failures:
* site `drake-linux-trusty-i-2017-04-26-c4.8xlarge`:
   * build `linux-gcc-nightly-coverage-matlab` - [Job 418](https://drake-jenkins.csail.mit.edu/view/Coverage/job/linux-gcc-nightly-coverage-matlab/418/) 
   * build `linux-gcc-nightly-coverage-matlab-open-source` - [Job 341](https://drake-jenkins.csail.mit.edu/view/Coverage/job/linux-gcc-nightly-coverage-matlab-open-source/341/)

Successes:
* site `drake-linux-trusty-i-2017-04-26-c4.8xlarge`:
    * build `linux-gcc-continuous-matlab-debug` - [Job 1419](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-gcc-continuous-matlab-debug/1419/)
    * build `linux-gcc-continuous-matlab-release` - [Job 2163](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-gcc-continuous-matlab-release/2163/)
    * ... others

PR #6016 (2017-05-04): Only run MATLAB mathematical program test if Python bindings are available
Fixes #6013. See also https://github.com/RobotLocomotion/drake-ci/commit/cc637d041d0b6e16fc303340d745589420021189. If the MATLAB bindings end up always deriving from the Python ones, we can simplify this.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6016)
closed 2017-05-04 by jamiesnape