RobotLocomotion/drake #6026: iiwa_ik_test fails on LLVM coverage builds

jamiesnape, 2017-05-04:
```
LLVM Profile Note: Set profile file path to "default.profraw" via default setting.
Running main() from gtest_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from testInverseKinematics
[ RUN      ] testInverseKinematics.SolveIkFromFk
Welding joint iiwa_joint_ee
Welding joint tool0_joint
Welding joint iiwa_joint_ee
Welding joint tool0_joint
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:78: Failure
Value of: (pos_diff.array() <= kUpperBound.array()).all()
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:79: Failure
Value of: (pos_diff.array() >= kLowerBound.array()).all()
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:82: Failure
Value of: std::cos(rot_diff) + kEpsilon >= std::cos(wp.rot_tol)
  Actual: false
Expected: true
[2017-05-04 04:55:14.553] [console] [warning] IK FAILED at max constraint relaxing iter: 11
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:78: Failure
Value of: (pos_diff.array() <= kUpperBound.array()).all()
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:79: Failure
Value of: (pos_diff.array() >= kLowerBound.array()).all()
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:82: Failure
Value of: std::cos(rot_diff) + kEpsilon >= std::cos(wp.rot_tol)
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:78: Failure
Value of: (pos_diff.array() <= kUpperBound.array()).all()
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:79: Failure
Value of: (pos_diff.array() >= kLowerBound.array()).all()
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:82: Failure
Value of: std::cos(rot_diff) + kEpsilon >= std::cos(wp.rot_tol)
  Actual: false
Expected: true
[2017-05-04 04:55:23.588] [console] [warning] IK FAILED at max constraint relaxing iter: 11
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:78: Failure
Value of: (pos_diff.array() <= kUpperBound.array()).all()
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:82: Failure
Value of: std::cos(rot_diff) + kEpsilon >= std::cos(wp.rot_tol)
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:78: Failure
Value of: (pos_diff.array() <= kUpperBound.array()).all()
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:79: Failure
Value of: (pos_diff.array() >= kLowerBound.array()).all()
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:82: Failure
Value of: std::cos(rot_diff) + kEpsilon >= std::cos(wp.rot_tol)
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:78: Failure
[... 179 lines cut ...]
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:79: Failure
Value of: (pos_diff.array() >= kLowerBound.array()).all()
  Actual: false
Expected: true
/home/ubuntu/workspace/linux-clang-nightly-coverage-open-source/drake/examples/kuka_iiwa_arm/dev/test/iiwa_ik_test.cc:82: Failure
Value of: std::cos(rot_diff) + kEpsilon >= std::cos(wp.rot_tol)
  Actual: false
Expected: true
[  FAILED  ] testInverseKinematics.SolveIkFromFk (257687 ms)
[----------] 1 test from testInverseKinematics (257688 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (257688 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] testInverseKinematics.SolveIkFromFk

 1 FAILED TEST
```

https://drake-cdash.csail.mit.edu/testDetails.php?test=12298860&build=574569
https://drake-cdash.csail.mit.edu/testDetails.php?test=12296003&build=574593
https://drake-cdash.csail.mit.edu/testDetails.php?test=12301779&build=574796

jamiesnape, 2017-05-12:
Also times out on various memcheck builds.

siyuanfeng-tri, 2017-05-12:
i volunteered @naveenoid to take a look.

PR #6090 (2017-05-12): SNOPT only IIWA IK test
I think this might tentatively resolve #6026

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6090)
closed 2017-05-12 by naveenoid