RobotLocomotion/drake #6039: monolithic pick and place bazel dependencies

RussTedrake, 2017-05-05:
i saw that the new Readme recommends running from the root directory, so I tried 
```
russt@ubuntu:~/drake-distro/drake/examples/kuka_iiwa_arm/dev/monolithic_pick_and_place$ bazel run :monolithic_pick_and_place_demo 
INFO: Found 1 target...
Target //drake/examples/kuka_iiwa_arm/dev/monolithic_pick_and_place:monolithic_pick_and_place_demo up-to-date:
  bazel-bin/drake/examples/kuka_iiwa_arm/dev/monolithic_pick_and_place/monolithic_pick_and_place_demo
INFO: Elapsed time: 0.238s, Critical Path: 0.00s

INFO: Running command line: bazel-bin/drake/examples/kuka_iiwa_arm/dev/monolithic_pick_and_place/monolithic_pick_and_place_demo
terminate called after throwing an instance of 'std::runtime_error'
  what():  drake::parsers::GetFullPath: ERROR: file_name "drake/examples/kuka_iiwa_arm/models/table/extra_heavy_duty_table_surface_only_collision.sdf" is not a file or does not exist.
```
I believe that the inability to launch via bazel run (from any directory) indicates missing dependencies (here the sdf files) in the bazel BUILD rules.

RussTedrake, 2017-05-07:
Also please mention the need to open a drake-visualizer (manually) in the README.md

(see the readme in drake/automotive for an alternative approach — which launches the viewer when the demo gets run).

naveenoid, 2017-05-11:
Fixed through #6056

closed 2017-05-11 by naveenoid