RobotLocomotion/drake #6052: Missing symbol '_ZdlPvm'

yaohuig, 2017-05-07:
When I run the example Quadrotor2D after installation, MATLAB shows following error:

------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------
Invalid MEX-file
'/home/gyh/Drake/drake-distro/drake/matlab/systems/trajectories/PPTmex.mexa64':
Missing symbol '_ZdlPvm' required by
'/home/gyh/Drake/drake-distro/drake/matlab/systems/trajectories/PPTmex.mexa64'.

Error in PPTrajectory (line 22)
        obj.mex_ptr = SharedDataHandle(PPTmex(obj.pp.breaks,
        obj.pp.coefs, obj.pp.order, obj.pp.dim));

Error in DynamicalSystem/simulate/makeSubTrajectory (line 167)
      traj = PPTrajectory(spline(t,y));

Error in DynamicalSystem/simulate (line 236)
      ytraj = makeSubTrajectory(t,y);

Error in DrakeSystem/simulate (line 494)
        [varargout{:}] = simulate@DynamicalSystem(obj,varargin{:});

Error in runLQR (line 19)
    xtraj=simulate(sys,[0 4],x0 + .99*y(:,i));
------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------

I followed the installation steps and my VTK version is 6. After installation, I run the addpath_drake.m in MATLAB and then run the example.
Is there any step I am missing? THank you!

hongkai-dai, 2017-05-07:
Can you make sure you libstdc++.so.6 is pointing to the right location? You can do so by opening a terminal, and do
```
$ ls -l MATLAB_PATH/sys/os/glnxa64/libstdc++.so.6
```
`MATLAB_PATH` is where you installed MATLAB. If it points to the libstdc++.so.6 coming with MATLAB, then you need to change the symbolic link, to point to the system libstdc++.so.6. You can do so by
```
$ cd MATLAB_PATH/sys/os/glnxa64
$ rm libstdc++.so.6
$ ln -s /usr/lib/gcc/x86_64-linux-gnu/YOUR_GCC_VERSION/libstdc++.so
```

yaohuig, 2017-05-08:
I followed the given command and it works! Thank you!

closed 2017-05-08 by hongkai-dai