RobotLocomotion/drake #6067: Fix some brittleness issues identified with RotationalInertia

amcastro-tri, 2017-05-10:
1.  Add static asserts to ensure we are using the lower portion of the underlying dense matrix. Per comment [here](https://reviewable.io/reviews/robotlocomotion/drake/5945#-Kjk5x7d8BPlefOZzSIo), a good idea is to have an enum with the triangular portion in use (as in original design).
2. Consider returning `Map` (or any other Eigen expression that allows a fixed-size stride of 4) to implement `get_moments()`. Per comment [here](https://reviewable.io/reviews/robotlocomotion/drake/5945#-Kjk1mYAzNT8KkmYZyWb).
3. Add a `rotational_inertia.cc` file explicitly instantiating in the most common Drake scalar types: double, AutoDiffXd, symbolic::Expression. Also, consider using SFINAE for compuational methods like `CalcPrincipalMomentsOfInertia()` that do not work on other types other than `double`. Document appropriately. 

cc'ing interested parties to chip in on this proposal. @sherm1, @edrumwri, @mitiguy, @jwnimmer-tri.

sherm1, 2017-05-10:
My 2 cents:
1. I do not understand how that helps anything. The class is defined to use the lower triangle -- all the methods should be implemented that way. How is that brittle?
2. Although it is a cool idea, I don't think it is a good idea to return a map with non-unit stride. That is asking for difficult compiler error messages downstream and doesn't seem justified in this non-performance critical method vs. simply copying out to a Vector3. If we were returning *mutable* access to the products of inertia it would make more sense, but that is a bad idea for other reasons!
3. Explicit instantiations: definitely. SFINAE: maybe wait until we have a need?

jwnimmer-tri, 2017-06-19:
All issues must have owners.  I'm assigning one arbitrarily.  (Fix it up if I'm wrong.)

jwnimmer-tri, 2023-05-23:
I'm not sure that after 6 years this issue is doing us any good.  My suggestion is to either close it now, or else convert to TODOs in the code at places where the code could be improved and close the issue.  Nobody working on the code will ever find the issue.  At least a TODO comment would mean that next time someone is working in that file, they might clean up one of the pieces as part of a related feature.

sherm1, 2023-05-23:
RotationalInertia has been improved a lot in six years. If there are still problems they should get fresh issues.

closed 2023-05-23 by sherm1