RobotLocomotion/drake #6082: Need to move the monolithic pick-and-place demo out of dev

naveenoid, 2017-05-11:
Multiple teams are using the KUKA IIWA Arm monolithic pick and place demo as a benchmark. It needs to move out of dev and into mainstream master. This issue tracks the PRs that need to land in order to achieve this.
- [ ] Move Iiwa_ik_planner -> add a suitable unit test.
- [ ] Move IiwaWsgPlantGeneratorsEstimatorsAndVisualizer
- [ ] Move ActionPrimitives (two or more PRs may be needed here).
- [ ] Move StateMachineAndPrimitives diagram.
- [ ] Move the demo itself (bikeshedding on the name needed!).

naveenoid, 2017-05-11:
Tagging @sammy-tri @kunimatsu-tri @SeanCurtis-TRI @siyuanfeng-tri @RussTedrake 
(People who might care about this move)

sammy-tri, 2019-02-14:
Closing because #10591 deleted the relevant code.

closed 2019-02-14 by sammy-tri