RobotLocomotion/drake #6086: Feature - optitrack driven pick and place demo (real robot)

naveenoid, 2017-05-12:
This issue tracks the various PRs needed to land an optitrack data driven pick and place demo in the real KUKA IIWA robot.

- [ ] Moving Average Filter (#6065)
- [ ] Optitrack LCM to pose translator (#6081)
- [ ] Substep 1 for the decoupled pick and place demo : WorldState needs to have an optitrack LCM as an option against the OracularStateEstimator
- [ ] Substep 2 : Pick and place demo to take in command line options to use optitrack LCM
- [ ] Substep 1 for monolithic pick and place demo : Create new demo diagram from LCM subscribers to optitrack
- [ ] Substep 2 : Pick and place demo to take in command line options to use optitrack LCM

naveenoid, 2017-05-12:
Tagging @sammy-tri

sammy-tri, 2019-03-04:
The pick and place demo has been removed.  I think it's safe to say there's no further outstanding work we're planning to do for it.

closed 2019-03-04 by sammy-tri