RobotLocomotion/drake #6089: Eigen3 not found in drake-distro

Jaeyoung-Lim, 2017-05-12:
I am trying to build a package called eigen_catkin using `catkin build` but the systems is trying to find eigen3 in a directory in `drake-distro/indlcude` which does not exist.

The error is as show below. 
```
Errors     << eigen_catkin:cmake /home/cosmonaut/catkin_ws/logs/eigen_catkin/build.cmake.007.log                                    
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:293 (message):
  catkin_package() absolute include dir
  '/home/cosmonaut/drake-distro/include/eigen3' does not exist
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:100 (_catkin_package)
  /home/cosmonaut/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:214 (catkin_package)
  CMakeLists.txt:117 (cs_export)

cd /home/cosmonaut/catkin_ws/build/eigen_catkin; catkin build --get-env eigen_catkin | catkin env -si  /home/cosmonaut/tools/cmake-3.5.2-Linux-x86_64/bin/cmake /home/cosmonaut/catkin_ws/src/eigen_catkin --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/cosmonaut/catkin_ws/devel/.private/eigen_catkin -DCMAKE_INSTALL_PREFIX=/home/cosmonaut/catkin_ws/install; cd -
```

Does anybody have an idea how to fix this?

Thank you

For reference, the link for eigen_catkin package is [here](https://github.com/ethz-asl/eigen_catkin)

naveenoid, 2017-05-12:
@liangfok : Would you perhaps know if we have a drake - eigen class when using catkin build? 
Also tagging @stonier

liangfok, 2017-05-12:
Drake downloads Eigen 3.3.3 during it's build process:

https://github.com/RobotLocomotion/drake/blob/2883a4b0b65b3a1cc494fa9708b64e14c1ff6c72/CMakeLists.txt#L34

I'm unsure why `eigen_catkin` is looking in Drake's subdirectory for `eigen3`. @Jaeyoung-Lim, can you post detailed replication steps?

Jaeyoung-Lim, 2017-05-12:
@liangfok The replication steps are as below
Thanks
```
cd ~/catkin_ws/src
git clone 
git clone https://github.com/ethz-asl/eigen_catkin.git
cd ..
catkin build eigen_catkin
```

Jaeyoung-Lim, 2017-05-12:
I have tried changing the CMakeLists.txt in the eigen_catkin package 
```
set(EIGEN_CATKIN_EIGEN_INCLUDE ${CATKIN_DEVEL_PREFIX}/include/eigen3)
```
to 
```
set(EIGEN_CATKIN_EIGEN_INCLUDE ${CATKIN_DEVEL_PREFIX}/externals/eigen)
```
to maybe guide it to find eigen in the drake directory as `drake-dsitro/include` doesn't exitst.  However, the same error looking for 'drake-distro/include/eigen3' appears

liangfok, 2017-05-13:
Using a fresh ROS workspace, I failed to replicate the problem described above, but ran into a different problem:

```
$ cd src
$ git clone git@github.com:catkin/catkin_simple.git
$ git clone git clone https://github.com/ethz-asl/eigen_catkin.git
$ git clone https://github.com/ethz-asl/eigen_catkin.git
$ cd ..
$ source /opt/ros/indigo/setup.bash
$ catkin init
$ catkin config
$ catkin build eigen_catkin
...
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:291 (message):
  catkin_package() absolute include dir
  '/home/liang/dev/drake_catkin_workspace/build/include/eigen3' does not
  exist
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:98 (_catkin_package)
  /home/liang/dev/test_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:214 (catkin_package)
  CMakeLists.txt:117 (cs_export)
```
Which version of Ubuntu and ROS are you using?

liangfok, 2017-05-13:
Oh, never mind, maybe the error I see is the same as the error you're observing. `find_package(Eigen3 QUIET)` is apparently finding it in a different workspace that contains Drake.

liangfok, 2017-05-13:
Running `catkin clean` on my drake-containing workspace fixed the problem. I believe this confirms that having a built workspace containing drake results in `eigen_catkin` finding the wrong eigen3 even when it's in a different workspace.

stonier, 2017-05-13:
@liangfok definitely and @Jaeyoung-Lim probably - sounds like you've built and sourced a drake containing catkin devel workspace and then tried to catkin build the eigen_catkin workspace afterwards. This means the drake containing workspace has become an underlay and as such, sets Eigen3 before eigen_catkin can get a look in.

1. Make sure you don't source any devel or install spaces in `.bashrc` and `.profile`. This just causes headaches when you are working with multiple workspaces.
2. Start from a fresh shell which hasn't sourced any other catkin devel or install space - your eigen_catkin only steps should be fine now

Note - I don't particularly like this behaviour of catkin tools. It's simple for someone who is only interested in building a single workspace. We built a different set of tools for catkin at Yujin which would only explicitly accept underlays to avoid these kind of problems.

Jaeyoung-Lim, 2017-05-16:
@liangfok @stonier Thank you for the response. However, I could not resolve the issue. 

@liangfok I have built drake using make and NOT as a catkin package. Thus, I cannot run `catkin clean ` in the drake catkin workspace. I am running ROS Indigo on Ubuntu 14.04

@stonier I have tried building only `eigen_catkin` in a fresh catkin workspace but the error is still not resolved

I will really appreciate if there are any suggestions in solving this issue

liangfok, 2017-05-16:
It's very strange how `eigen_catkin` can find `drake` even when `drake` is not built using `catkin`. Have you tried running `make clean` in your `drake` directory?

``
cd /home/cosmonaut/drake-distro/build
make clean
```

stonier, 2017-05-16:
Resolved over a quick hangout. It was as suspected, an old catkin workspace build with drake penetrating his new via catkin's overlay mechanism.

closed 2017-05-16 by stonier