RobotLocomotion/drake #6095: Error while executing addpath_drake

GPrathap, 2017-05-15:
I followed drake installing using [make](http://drake.mit.edu/from_source.html#build-with-make). The build was successful. But then I tried to test drake installation in Matlab using [this](http://drake.mit.edu/from_source.html#test-your-installation).  

But I am getting this error. 

```
Error using addpath_drake (line 20)
The Drake build olsutputs are not on the MATLAB path, and could not be auto-detected. Please add the build output directory "matlab" to the path, then re-run addpath_drake.
```

I have checked `drake-distro/build/install/matlab` directory and I can see only these m files. `addpath_sedumi.m  addpath_spotless.m  rmpath_sedumi.m  rmpath_spotless.m`. 

Then I tried to add  addpath_spotless.m and it works fine. Can someone explain is this related build issue or something else.

Environment: 16.04.1-Ubuntu, Matlab16a

hongkai-dai, 2017-05-15:
Could you make sure that drake is built with MATLAB? You can do the following to check if MATLAB was detected when compiling drake
```
$ cd drake-distro/build
$ ccmake ..
```
When I do this on my machine, it shows the variables DISABLE_MATLAB, MATLAB_EXECUTABLE, Matlab_ROOT_DIR, etc, are set to some values, as in the following screenshot. Can you make sure that cmake detects your MATLAB?
[image]

GPrathap, 2017-05-15:
Yes, This is my configuration.  
[image]

Anyway thanks for the prompt response. If you have a build can you give me the what files should be in the `drake-distro/build/install/matlab`.

hongkai-dai, 2017-05-15:
The files in my `drake-distro/build/install/matlab` are
[image]

hongkai-dai, 2017-05-15:
BTW, the files in `drake-distro/drake/matlab` are
[image]

GPrathap, 2017-05-16:
Files in the `drake-distro/drake/matlab` are very similar to what you have. But most of them did not copy to `drake-distro/build/install/matlab`. I will try to install Matlab16b or Matlab17a then will rebuild the drake. Will let you know the result.  Thanks for your time on this.

jwnimmer-tri, 2017-06-19:
All issues must have owners.  I'm assigning one arbitrarily.  (Fix it up if I'm wrong.)

closed 2018-09-27 by jwnimmer-tri